OpenTracksApp/OpenTracks #1885: Crash when record started

john-parker, 2024-04-04:
### App information
* ID: de.dennisguse.opentracks
* Version: 5725 v4.12.0

### Device information
* Brand: google
* Device: sargo
* Model: Pixel 3a
* Id: UQ1A.240205.004
* Product: sargo

### Firmware
* SDK: 34
* Release: 14
* Incremental: 24504010

### Cause of error
```java
Exception in thread "main": android.app.RemoteServiceException$ForegroundServiceDidNotStartInTimeException: Context.startForegroundService() did not then call Service.startForeground(): ServiceRecord{8058ba5 u0 de.dennisguse.opentracks/.services.TrackRecordingService}
	at android.app.ActivityThread.generateForegroundServiceDidNotStartInTimeException(ActivityThread.java:2129)
	at android.app.ActivityThread.throwRemoteServiceException(ActivityThread.java:2100)
	at android.app.ActivityThread.-$$Nest$mthrowRemoteServiceException(Unknown Source:0)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2394)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:205)
	at android.os.Looper.loop(Looper.java:294)
	at android.app.ActivityThread.main(ActivityThread.java:8248)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
```

john-parker, 2024-04-04:
FYI, 'm using CalyxOS version 5.4.1, Android version 14. I have version v4.12.0/v4.12.0 Version code 5725 and have tried uninstalling and re-installing OpenTracks. Please let me know if there is any other information that may be useful.

dennisguse, 2024-04-05:
1. What were you doing? (aka how to reproduce)
2. Is this always happening?

john-parker, 2024-04-05:
> What were you doing? (aka how to reproduce)

- Press Record. 

    Is this always happening?

- Yes. Last correct behaviour was Monday April 1. In last couple of days pressing record brings up the OpenTracks crashed screen (and later the Android "OpenTracks keep stopping - App info/Close App"). Note that after crashing a track is added to the list of tracks with zero distance, zero time, zero speed etc.

dennisguse, 2024-04-05:
Okay.
Was there an update of OpenTracks?
Can you try the previous version?

john-parker, 2024-04-05:
I think I did update it in the last few days. I have tried all the versions currently on f-droid:
Version: 5725 v4.12.0
Version: 5705 v4.11.3
Version: 5683 v4.11.2
All with the same result - crash upon start of recording.

dennisguse, 2024-04-06:
That sounds like it is a problem outside of OpenTracks. 
Did you get an update of the Android OS?

sgofferj, 2024-04-06:
I just noticed that my automatic recording doesn't work since last Wednesday and I suspect it might be the same issue.
I'm using nightly 4.12.0-4-g4bfd65bb6 Version Code 5730 on Android 12.
Apparently when tasker sends the intent to start recording, OT does start the track and dies. The track itself with all Metadata from the intent call is in the list  but it's empty.
I'm not sure which version still worked but the last successful recording was on 2024-04-03 15:21:49 EEST.

squishyu, 2024-04-10:
I have the same issue. Since a recent update (somewhere at the start of April) OpenTracks seems to soft-crash when starting an activity (in my case through GadgetBridge)

When I start the activity and switch to OpenaTracks I get a white screen
[image]

When I the kill and reopen the app, there's a new entry in the activities but with no data (AVG speed is NaN)
[image]

john-parker, 2024-04-11:
> That sounds like it is a problem outside of OpenTracks. Did you get an update of the Android OS?
I hae Android 14. The firmware, vendor and kernal are all from 2022/2023 and the last Android security update was for the platform on Feb 5, 2024. So these are date too early to be the issue.

I believe I did have an OS update in the last couple of weeks (I can't find the date). Here is the change log:

- CalyxOS 5.4.1
- Feature update

**Features and bug fixes**

- Chromium: 122.0.6261.90
- Work profile work hours feature - Set working hours for your work profile
- Settings -> Accounts -> Work -> Work profile settings
- usb: Show charging notification only if power brick is disconnected
- Fix unknown sources app install confirmation
- Fix uninstall for all users
- Fix headline font in various places, avoid inconsistencies
- Quick settings: Allow long-pressing on mobile network
- Various updates for included apps
- Updated translations for all apps, and the OS

john-parker, 2024-04-19:
@dennisguse thank you, I will try and report back.

P.S. I enjoy this app. I have used it since the Google days and this incarnation is better. That I can use it locally and privately and this fits with my view of how apps should work. It is simple and well-designed. Thank you to the team developing and maintaining it.

DrOuille9, 2024-04-21:
Hey!
Same problem here, I usually having it several times, when clicking on the big recording button :
https://anonpaste.pw/v/2f785a4c-f7a9-4f97-8dcc-db4db8bb9c9d#3mqHCXXO0Vye8AmixUzum2ZuKAA08nOXZ4QKC_N4Fk4

But I have noticed the crash is coming after clicking on the little "target" icon (I think it's to force the GPS waking up ? With the little dot blinking inside, or no dot if no GPS...)
If no click on this "target" icon, usually no crash when starting to record.

Same for you @john-parker

DrOuille9, 2024-04-22:
4.12.2 release looks like fixing this issue :)
Tried 1 time without crash, it needs more test ;)

Thx for this ! Good work !

john-parker, 2024-04-24:
@dennisguse thank you, I will try and report back.

P.S. I enjoy this app. I have used it since the Google days and this incarnation is better. That I can use it locally and privately and this fits with my view of how apps should work. It is simple and well-designed. Thank you to the team developing and maintaining it.

> Hey! Same problem here, I usually having it several times, when clicking on the big recording button : [...]

No. "target icon" makes no difference.

john-parker, 2024-04-24:
> @john-parker I don't know a good idea how to continue here. Some ideas:
> 
> 1. You could try older versions of OpenTracks (not only the ones on F-Droid)
> 2. You check if there is a prior error resulting in this exception (crash and then the reported crash due to the timeout).
> 
> @squishyu can you try to get a stack trace? The screenshots itself are not too helpful ;) And this issue is not about the crash due to start via the Public API, but via the UI istelf. I guess a separate issue may be helpful.

Success! 

Reverting to 4.11.1 provided a fix for me. 

Now OpenTracks works as expected and does not crash immediately upon starting a recording.

dennisguse, 2024-05-01:
@john-parker can you do the brute force approach by upgrading one release at a time?
Or did you downgrade this way already and v4.11.1 is first one that worked again for you?

john-parker, 2024-05-01:
@dennisguse I am not sure about what you mean by the first question, but I did indeed delete each release and reinstall the next available downgrade and v4.11.1 was the first one that worked. (P.S. I don't know if there is ane easier way but I took Versions: v4.12.0; v4.11.3; and v4.11.2 from F-Droid and v4.11.1 from your repo - https://github.com/OpenTracksApp/OpenTracks/releases -and then built the .apk file from there.

dennisguse, 2024-05-03:
@john-parker you did exactly what I asked :)
Since you can compile yourself and I know it is quite some effort:
Can you start v4.11.1 (which actually is slightly broken) and check which of the commits (that has code/xml changes) broke it for you.
The first functional change I am aware of if this is dfdbc373ff90c37c8fb5fab396f3c470eee65772 (released in v4.12.1).

The rest is related to Public API or translations...

john-parker, 2024-05-13:
My apologies. I ain't no programmer. I think I built the apk for 4.11.1 out of sheer dumb luck. I can't remember what I did in Android Studio to get the build to work. If you could give me a bit more direction (actually a lot more) I might be able to help.

When I "start v4.11.1"  in Android Studio and on the phone everything works fine so I am not sure what is involved in "check which of the commits (that has code/xml changes) broke it for you"?

dennisguse, 2024-09-08:
@john-parker is this still an issue for you?

closed 2024-12-19 by dennisguse