OpenTracksApp/OpenTracks #1889: Compute distance taking into consideration hill

xfarrow, 2024-04-10:
**Is your feature request related to a problem? Please describe.**
OpenTracks seems it computes distances based on the relation between two points on a 2D map. This implies that, for example, when walking uphill or downhill, the actual distance is greater than the one that is shown.

**Describe the solution you'd like**
It'd be amazing if OpenTracks would somehow know the inclination I'm walking on.

**Describe alternatives you've considered**

**Additional context**
Thank you for considering my request. I don't know whether it can be done though.

dennisguse, 2024-04-11:
Why?
If your phone has a barometer, OpenTracks measure the gain. 

What is the benefit of merging both into distance?

xfarrow, 2024-04-11:
Hi @dennisguse thank you for your response. 

I will try to explain myself better. Roughly, the distance from the bottom of Mt. Everest and its top, looked from a 2D satellite image, is 10km. If I climbed the Mt Everest in a straight line, however, due to its three-dimensional nature, I'll walk more.

Thinking to an extreme would be climbing a ladder, where both the start and end point coincide if I look at it from above on a 2D map, so the app would think I walked 0m, but the distance I travel is not 0m, notably it is the height of the ladder.

However, I have no idea whether this can be addressed. I

dennisguse, 2024-04-12:
In my understanding, you have the elevation gain for this case.
This exactly shows you the how much you climbed/walked uphill.

Do you know how other GPS trackers like Garmin are doing this?

xfarrow, 2024-04-12:
Unfortunately I don't know how they work, but you're right, we still have the elevation from the app :)

closed 2024-05-01 by dennisguse