OpenTracksApp/OpenTracks #1892: Error when starting to track with GPS

JonKo314, 2024-04-14:
### App information
* ID: de.dennisguse.opentracks
* Version: 5730 v4.12.1

### Device information
* Brand: Fairphone
* Device: FP3
* Model: FP3
* Id: TQ3A.230901.001
* Product: lineage_FP3

### Firmware
* SDK: 33
* Release: 13
* Incremental: eng.root.20240323.025914

### Cause of error
```java
Exception in thread "main": java.lang.RuntimeException: SensorManager cannot be started twice; stop first.
	at de.dennisguse.opentracks.sensors.SensorManager.start(SensorManager.java:67)
	at de.dennisguse.opentracks.services.handlers.TrackPointCreator.start(TrackPointCreator.java:47)
	at de.dennisguse.opentracks.services.TrackRecordingService.startSensors(TrackRecordingService.java:211)
	at de.dennisguse.opentracks.services.TrackRecordingService.startRecording(TrackRecordingService.java:192)
	at de.dennisguse.opentracks.services.TrackRecordingService.startNewTrack(TrackRecordingService.java:172)
	at de.dennisguse.opentracks.TrackListActivity.lambda$onCreate$6(TrackListActivity.java:184)
	at de.dennisguse.opentracks.TrackListActivity.$r8$lambda$zZE3EWVw__RyZbejvjKvasa-kHM(Unknown Source:0)
	at de.dennisguse.opentracks.TrackListActivity$$ExternalSyntheticLambda4.onConnected(D8$$SyntheticClass:0)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.lambda$executeForeground$2(TrackRecordingServiceConnection.java:169)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection$$ExternalSyntheticLambda0.onConnected(D8$$SyntheticClass:0)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.setTrackRecordingService(TrackRecordingServiceConnection.java:135)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.-$$Nest$msetTrackRecordingService(Unknown Source:0)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection$1.onServiceConnected(TrackRecordingServiceConnection.java:62)
	at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:2188)
	at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:2221)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7924)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
```

### Additional Information
* I opened the app, pressed the GPS button in the lower left and then the red record button on the lower right.
* Happens consistently even after restarting the phone.

dennisguse, 2024-04-14:
Duplicate of #1887.
Is fixed in v4.12.2 - will be on F-Droid in a few days.

closed 2024-04-14 by dennisguse

JonKo314, 2024-04-14:
Oh, sorry, I didn't check the closed issues. Thanks though!

dennisguse, 2024-04-14:
No worries - better one report to many than none 😁