OpenTracksApp/OpenTracks #1897: Periodic crash when resuming stopped track

SamWhited, 2024-04-20:
Sometimes when I resume a stopped track OpenTracks crashes. If I do it again, it normally works. The crash appears to be intermittent. Crash information copied below:

```
### App information
* ID: de.dennisguse.opentracks
* Version: 5730 v4.12.1

### Device information
* Brand: asus
* Device: ASUS_I006D
* Model: ASUS_I006D
* Id: TQ3A.230901.001
* Product: WW_I006D

### Firmware
* SDK: 33
* Release: 13
* Incremental: f1f5f2db06

### Cause of error
```java
Exception in thread "main": java.lang.RuntimeException: SensorManager cannot be started twice; stop first.
	at de.dennisguse.opentracks.sensors.SensorManager.start(SensorManager.java:67)
	at de.dennisguse.opentracks.services.handlers.TrackPointCreator.start(TrackPointCreator.java:47)
	at de.dennisguse.opentracks.services.TrackRecordingService.startSensors(TrackRecordingService.java:211)
	at de.dennisguse.opentracks.services.TrackRecordingService.startRecording(TrackRecordingService.java:192)
	at de.dennisguse.opentracks.services.TrackRecordingService.resumeTrack(TrackRecordingService.java:185)
	at de.dennisguse.opentracks.TrackRecordedActivity.lambda$onOptionsItemSelected$2(TrackRecordedActivity.java:205)
	at de.dennisguse.opentracks.TrackRecordedActivity.$r8$lambda$2nGrZpROckbHDFXbMtLjiI8qENA(Unknown Source:0)
	at de.dennisguse.opentracks.TrackRecordedActivity$$ExternalSyntheticLambda2.onConnected(D8$$SyntheticClass:0)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.lambda$executeForeground$2(TrackRecordingServiceConnection.java:169)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection$$ExternalSyntheticLambda0.onConnected(D8$$SyntheticClass:0)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.setTrackRecordingService(TrackRecordingServiceConnection.java:135)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.-$$Nest$msetTrackRecordingService(Unknown Source:0)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection$1.onServiceConnected(TrackRecordingServiceConnection.java:62)
	at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:2188)
	at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:2221)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7924)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
```

dennisguse, 2024-04-20:
Similar to #1887 - should be fixed in v4.12.2

PS/ the error indicates that the sensors were started.
Did you do this or were they not stopped properly?

closed 2024-04-20 by dennisguse

SamWhited, 2024-04-20:
> PS/ the error indicates that the sensors were started.
> Did you do this or were they not stopped properly?

Ahh, yes, that was it. It looks like if I hit the "start GPS" button and *then* resume the recording I can reproduce this. I'm not sure if that's what I did when I got this log or if it was something else, but that does seem to reliably reproduce the problem.