OpenTracksApp/OpenTracks #1898: strange numbers recorded

mega-stoffel, 2024-04-21:
Hej there!

Today's cycling recording looks strange.

[image]

**Describe the bug**
It just happened today, after succesfully recording about 20 tracks on this device. Maybe related to stops/breaks inside without/bad GPS coverage.

**To Reproduce**
not sure, if it's reproducible. I'll watch out for it.

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: Pixel 6
 - OS: GrapheneOS 2024040900
 - OpenTracks version: 4.12.2
 - OpenTracks commit id (only for nightly builds): [e.g. 4b333bb]

dennisguse, 2024-04-22:
Did you use GPS or some Bluetooth sensor?
We had an overflow in the distance parsing of Bluetooth sensors that looked similar.

mega-stoffel, 2024-04-22:
just saw it again.
I'm just using the phone's GPS, no Bluetooth devices, Bluetooth is even switched off completly.
Didn't find the time to analyze the tracks, yet.

dennisguse, 2024-04-22:
Under these circumstances; can you send me a track (preferably via email)?

mega-stoffel, 2024-04-22:
just updated to GrapheneOS 2024042100 and it's still happening. Recorded the same route additionally with OSMand, and there I have a correct number for the kilometers.
So I assume, it's related to OpenTracks, maybe introduced in the latest version/patch, since I've never seen it before?

mega-stoffel, 2024-04-22:
I'm afraid, I can't send you mails?

> The mail system
> 
> <dennis.guse@.....>: TLS is required, but host
>     mailbox.alum....[130.149.5.95] refused to start TLS: 454 TLS
>     currently unavailable

ScriptForAll, 2024-04-23:
I found a common cause.
Whenever I forget to finish the recording and enter a building these kind of numbers appear.

Maybe it's creating a position while the GPS doesn't give one?

filipsobol, 2024-04-23:
I've been using OT for about 2 years, but this problem started recently, probably after one of the recent updates.

I believe @ScriptForAll might be onto something. Last week, I had two activities where the app recorded distances of nearly 50k km and 6k km.

After that, I recorded an activity where I didn't enter a building and it was fine. Then, I logged one where I entered a building for a short period of time, and the app recorded 43km.

dennisguse, 2024-04-24:
Please send me a track (as KML) to opentracks-play@dennisguse.de 

You can also check yourself:
1. How does the distance chart look like
2. Plot track on a map (e.g. OSMDashboard): does it look okay or are there outliers?
3. In the file (best KML): Are there any outlying GPS locations (likely at the end)
3. In the file (best KML): Are there more points included than in previous tracks?

The recent change that may be related: processing if GPS data was modified, so it is handled similarly to all other sensor data. 

PS as this is already live (and more than one person affected) this is urgent.

Grimtown, 2024-04-25:
Can confirm.
[image]

Nokia XR20
Android 13
OpenTracks v4.12.2, latest update

dennisguse, 2024-04-25:
The issue is that when **idle timeout** is reached, these coordinates are inserted:
```
<when>2024-04-22T23:02:48.425+02:00</when>
<coord>0.0 0.0</coord>
```

**Expectation:** it should be `<coord />` aka null values.
Introduced in dfdbc373ff90c37c8fb5fab396f3c470eee65772 (as side effect).
**Affected version:** 
* v4.12.0
* v4.12.1
* v4.12.2

**Workaround:** disable idle timeout.

@mega-stoffel thanks!

dennisguse, 2024-04-27:
Anybody who has some time today: can you check if this PR solves the issue for you?
I added a test for this, but it would be good to check if this actually fixes the problem (and not adds a regression).
Release is latest scheduled for Sunday.

https://github.com/OpenTracksApp/OpenTracks/pull/1903

Grimtown, 2024-04-27:
> Can confirm. ![image](https://private-user-images.githubusercontent.com/87585085/325697331-b42398b8-a33b-4470-96f9-819bdc60b5c4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQyMzcxOTYsIm5iZiI6MTcxNDIzNjg5NiwicGF0aCI6Ii84NzU4NTA4NS8zMjU2OTczMzEtYjQyMzk4YjgtYTMzYi00NDcwLTk2ZjktODE5YmRjNjBiNWM0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI3VDE2NTQ1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY0NWYyODRlMjY3MTQ4OThkZjM5YWI4MTU3YjM3Y2Y0ZWFhNTRkMTRjODM4YTVmYzYzNjk5ZjEwYTY0YjhlOWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bcYermHQM5wBXMg4AHKftVXsDV1EYmX5q9aHxcolqcM)
> 
> Nokia XR20 Android 13 OpenTracks v4.12.2, latest update
> 
> Map looks okay, no outliers.

Addendum: Happened again today, didn't apply the workaround, but noticed something else.
Both times this happened the "total time" and "moving time" was exactly the same despite the breaks I took, which should pause the "moving time".

closed 2024-04-28 by dennisguse

dennisguse, 2024-04-28:
Released: https://github.com/OpenTracksApp/OpenTracks/releases/tag/v4.12.3

Humanimaliberation, 2024-05-19:
Sure the update to v4.12.3 solved the issue?

Happened again to me Yesterday and in the night from Yesterday to today around 1 am. 
Distance jumped after idle time to >11.000 km.

Checked my OpenTracks Version Just now, but it shows 4.12.3 as version. 
I did Not manually Update anything since the record Yesterday. Also checked the Version in F Droid and it says that 4.12.3 is installed and was added (to F-Droid Store I guess) at 03.05.24.
So I assume I did have 4.12.3 installed when the issue happened Yesterday.

Anyone know what caused that?

Phone:
Doogee S89
Android 12
Kernel-version 
4.19.191-g267f96cf84e7-dirty
#1 Tue Feb 21 12:45:25 CST 2023
Build Number 
DOOGEE-S89-EEA-Android12.0-20230221

commit 37a53dcf (2024-04-26): Bugfix: overestimated distance.

A combination of idle trackpoint and stale GPS data, resulted in recording lat=0.0 and lng=0.0; and thus inflating distance as well as GPS-based speed.

Fixes #1898.

Introduced in dfdbc373ff90c37c8fb5fab396f3c470eee65772
PR #1903 (2024-04-28): Bugfix: overestimated distance.
A combination of idle trackpoint and stale GPS data, resulted in recording lat=0.0 and lng=0.0; and thus inflating distance as well as GPS-based speed.

Fixes #1898.

Introduced in dfdbc373ff90c37c8fb5fab396f3c470eee65772
commit e590e4e0 (2024-04-28): Bugfix: overestimated distance.

A combination of idle trackpoint and stale GPS data, resulted in recording lat=0.0 and lng=0.0; and thus inflating distance as well as GPS-based speed.

Fixes #1898.

Introduced in dfdbc373ff90c37c8fb5fab396f3c470eee65772