OpenTracksApp/OpenTracks #1900: Battery drain Sensors running background

pamatika, 2024-04-22:
**Describe the bug**
After last update tracker is keeping location / gps active all the time causing battery drain 16% 9 hours.

**To Reproduce**
1. After updating to latest version battery drain started. Also location notification for tracker popping up from time to time.

**Technical information**
 - Device: Samsung A34
 - OS: latest
 - OpenTracks version: v4.12.2

dennisguse, 2024-04-23:
How can I reproduce this?

pamatika, 2024-04-28:
So this happened after updating to the latest version. App kept on trying to get a fix on location. Killing the app didn`t help. But restart of the phone helped. After restart I have not seen same behaviour. So it has to be some clitch in installation...or..

dennisguse, 2024-04-28:
Thanks for the feedback. 
Let's close this issue for now and if it reappears, please open it again.

closed 2024-04-28 by dennisguse