OpenTracksApp/OpenTracks #1902: Samsung galaxy s10 not tracking in Background

FirstWithThisName, 2024-04-24:
**Please describe the problem and how to reproduce it:**

...

**Checklist**

- [x] Consulted [https://dontkillmyapp.com/]
- [x ] Power management: battery optimization is disabled for OpenTracks

**Technical information**

- Device: [Samsung Galaxy s10]
- OS: 12
- OpenTracks version: 4.12.0

OT only records if the screen is on.
I only can allow using the location while app usage.
Other app have the option to always allow using the location.

dm1cael, 2024-04-25:
Could #1653 be related?

dennisguse, 2024-04-28:
Might be #1653. 

If the screen is off, some Android devices might enforce permission to access the location in the background. 
And OpenTracks is not (yet) requesting this permission.

Can you check if it works, if the screen is on?

dennisguse, 2024-04-28:
How do you start OpenTracks?

FirstWithThisName, 2024-04-28:
> [Verfolgungsbild][https://github.com/notifications/beacon/AJT3YER7RCTMHWVJGZIRR7LY7VLSTA5CNFSM6AAAAABGXUSCI6WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT4CNO5A.gif]

Oh, yes. Sorry for the delayed answer.
With the screen unlocked it works.
#1653 seams to be the problem

FirstWithThisName, 2024-04-28:
> How do you start OpenTracks?

With the red button at the lower right corner

dennisguse, 2024-04-28:
That's a different problem than. 

PERMISSION_LOCATION_BACKGROUND is not required if started via the UI. 
In this case: it is usually a misconfiguration with power saving - e.g. something like Android OS optimizing for low power consumption.

FirstWithThisName, 2024-04-28:
I set it to not optiomize in the app settings but, my phone maybe been in energie saving mode. I will test tomorrow if switching of the energie saving mode fixed the issue.

FirstWithThisName, 2024-05-31:
Oh, sorry it seams like the e-mail resonse didn't work.

Disableing the engerie save mode fixed it.

Thanks for your help!

closed 2024-05-31 by FirstWithThisName