OpenTracksApp/OpenTracks #1904: When starting OpenTracks via GadgetBridge no data is being recorded

sebastianha, 2024-04-26:
When starting a new recording with my Mi Band 7 via Gadgetbridge a new recording is being started with opentracks but directly after the start stopped and no data is being recorded.

I know from old recordings that is worked until at least 2024-03-02, I just discovered that all other recordings are empty since then.

This is only the case when started via Gadgetbridge (via intent?), normal recording works. I checked all the settings, they seem to be correct.

...

**Checklist**

- [x] Consulted [https://dontkillmyapp.com/]
- [X] Power management: battery optimization is disabled for OpenTracks

**Technical information**

- Device: Pixel 7
- OS: GrapheneOS
- OpenTracks version: 4.12.2

dennisguse, 2024-04-28:
Might be #1653.

If the screen is off, some Android devices might enforce permission to access the location in the background.
And OpenTracks is not (yet) requesting this permission.

Can you check if it works, if the screen is on?

sebastianha, 2024-04-29:
Also does not work with screen on. I see that only Gadgetbridge is accessing Location and on Opentracks a new record is being created but directly stopped with no data.

dennisguse, 2024-04-29:
Can you check if there is a crash happening (check logcat)?

sebastianha, 2024-04-30:
I cannot see any crash, also when the app is open in foreground it stays open.
I do see a short popup at the top sometime when the app is not open and recording starts.

[image]

dennisguse, 2024-04-30:
Can you start a recording (that has data) via the UI?
And the notification is only shown briefly or does it remain in the notification bar (assuming OpenTracks has a the notification permissoin)?

sebastianha, 2024-04-30:
Yes, with the UI it works.
The notification is only visible for a split second.

MagiX13, 2024-05-01:
I have the same issue when starting to track via intents (I have a device that I use only for sport activities and use tasker to start tracking once the connection is established). It creates a track of 0 seconds and that's all. Manually starting to track works without issues. 
I'm also running on GrapheneOS, but with a different device. The issue started occuring somewhere between 9 and 18 March. I can't say if this comes from 4.12 that I just installed or the new Graphene [version](https://grapheneos.org/releases#2024031400) that came out during that time frame.
The GrapheneOS logs for the app don't show anything suspicious in my view, but happy to share if needed.

sebastianha, 2024-05-01:
> The GrapheneOS logs for the app don't show anything suspicious in my view, but happy to share if needed.

Same here

arjan-s, 2024-05-06:
Came here to report this exact issue, glad to find it's already known!

sgofferj, 2024-05-08:
I have the same issue and I observed the following:

- When I open OpenTracks, GPS is turned off (struck out icon)
- When I start a recording from the UI, GPS is turned on
- When I stop the recording in the UI, GPS is turned off again
- When I try to start a recording via intent, it doesn't record anything
- When I turn on GPS in the UI and then send the intent, OT records properly but turns off GPS after recording

Those issues did not occur before 03.04.2024. I'm using nightly via F-Droid on Andorid 12.

Just-Explode, 2024-08-16:
This also happens to me on either Android 14 (SM-A1560) for using Automate/App Start Block or adb shell intent ,or Android 11 (Waydroid) for using adb shell intent. I'm using v4.13.0 on both machine
I'm not sure about how this app works and I don't have much experience on coding Android app. But I wanna make a guess that maybe this bug is caused because it asks for location permission with "Allow only while using the app" instead of "All the time". Just my humble opinion. Maybe I was wrong, idk. I don't know how to touch these code yet.

Also I did 2 logcats on Waydroid, one is for "Activating GPS in main UI by touching the GPS button then run the intent", the other is "Start the main UI without touching the GPS button then run the intent". I use fake GPS to emulate GPS on Waydroid. And it can record on both machine by pressing the recording button in the main UI.

Here's "Activating GPS in main UI by touching the GPS button then run the intent" logcat:
```
08-16 13:48:54.090    84    84 E SurfaceFlinger: [Sprite#0] rejecting buffer: bufWidth=31, bufHeight=39, front.active_legacy.{w=35, h=35}
08-16 13:48:54.206 12664 12664 I TrackRecordingServiceConnection: Binding the service.
08-16 13:48:54.222 12664 12664 D TrackRecordingService: Create
08-16 13:48:54.222 12664 12664 I TrackRecordingService: new status null -> RecordingStatus[trackId=null]
08-16 13:48:54.224 12664 12664 I TrackRecordingServiceConnection: Connected to the service: de.dennisguse.opentracks.services.TrackRecordingService$Binder@f80aee1
08-16 13:48:54.224 12664 12664 I TrackRecordingServiceConnection: Connected to the service: de.dennisguse.opentracks.services.TrackRecordingService$Binder@f80aee1
08-16 13:48:54.225 12664 12664 I TrackRecordingService: tryStartSensors
08-16 13:48:54.225 12664 12664 I TrackRecordingService: startSensors
08-16 13:48:54.225 12664 12664 I SystemUtils: Acquiring wake lock.
08-16 13:48:54.226 12664 12664 I SensorDataSet: Setting aggregator de.dennisguse.opentracks.sensors.sensorData.AggregatorGPS to sensorAddress=internal data=null
08-16 13:48:54.227 12664 12664 I GPSManager: Register for location updates de.dennisguse.opentracks.services.TrackRecordingService@d826e3a
08-16 13:48:54.228 12664 12664 I TrackRecordingService: newGpsStatus: 2131820780
08-16 13:48:54.229 12664 12664 I SensorDataSet: Setting aggregator de.dennisguse.opentracks.sensors.sensorData.AggregatorBarometer to null
08-16 13:48:54.229 12664 12664 W BluetoothConnectionManager: Bluetooth not enabled.
08-16 13:48:54.231   252  5295 I ActivityManager: Killing 11478:com.google.android.gms.unstable/u0a144 (adj 995): empty for 1810s
08-16 13:48:54.229 12664 12664 I chatty  : uid=10176(de.dennisguse.opentracks) identical 3 lines
08-16 13:48:54.229 12664 12664 W BluetoothConnectionManager: Bluetooth not enabled.
08-16 13:48:54.232   252   280 W libprocessgroup: Failed to open process cgroup uid 10144 pid 11478: Permission denied
08-16 13:48:54.232   252   280 E libprocessgroup: Error encountered killing process cgroup uid 10144 pid 11478: Permission denied
08-16 13:48:54.240    62    62 I Zygote  : Process 11478 exited due to signal 9 (Killed)
08-16 13:48:54.293   252 12591 I system_server: oneway function results will be dropped but finished with status OK and parcel size 4
08-16 13:48:54.293   252 12591 I system_server: oneway function results will be dropped but finished with status OK and parcel size 4
08-16 13:48:54.294   252   267 I system_server: oneway function results will be dropped but finished with status OK and parcel size 4
08-16 13:48:54.432   252   252 D NotificationService: 0|de.dennisguse.opentracks|123|null|10176: granting content://settings/system/notification_sound
08-16 13:48:54.433   252   252 I chatty  : uid=1000 system_server identical 2 lines
08-16 13:48:54.433   252   252 D NotificationService: 0|de.dennisguse.opentracks|123|null|10176: granting content://settings/system/notification_sound
08-16 13:48:54.434   252   252 W NotificationHistory: Attempted to add notif for locked/gone/disabled user 0
08-16 13:48:54.436   129   675 D NuPlayerDriver: NuPlayerDriver(0xf7bc0600) created, clientPid(382)
08-16 13:48:54.440   129 12999 D GenericSource: FileSource remote
08-16 13:48:54.442   382   382 W NotifBindPipeline: Row is not set so pipeline will not run. notif = 0|de.dennisguse.opentracks|123|null|10176
08-16 13:48:54.445   127   391 E Utils   : did not find channel-count and/or sample-rate
08-16 13:48:54.446   127   391 E Utils   : did not find channel-count and/or sample-rate
08-16 13:48:54.447   382   382 W NotifBindPipeline: Row is not set so pipeline will not run. notif = 0|de.dennisguse.opentracks|123|null|10176
08-16 13:48:54.451   252 12591 I MediaFocusControl: requestAudioFocus() from uid/pid 10132/382 clientId=android.media.AudioManager@68d418 callingPack=com.android.systemui req=3 flags=0x0 sdk=30
08-16 13:48:54.452   382   382 W NotifBindPipeline: Row is not set so pipeline will not run. notif = 0|de.dennisguse.opentracks|123|null|10176
08-16 13:48:54.456   382   382 W NotifBindPipeline: Row is not set so pipeline will not run. notif = 0|de.dennisguse.opentracks|123|null|10176
08-16 13:48:54.474   382   382 W TouchableRegionManager: onHeadsUpPinnedModeChanged
08-16 13:48:54.475   382   382 D StatusBar: disable<e i a s b h r c s > disable2<q i n >
08-16 13:48:54.483   382   382 D StatusBar: disable<e i a s b h r c s > disable2<q i n >
[... cut ...]
08-16 13:50:31.972   129   675 W AMessage: failed to post message as target looper for handler 0 is gone.
08-16 13:50:31.973   252  5295 I MediaFocusControl: abandonAudioFocus() from uid/pid 10132/382 clientId=android.media.AudioManager@5946025
08-16 13:50:33.475   382   382 W TouchableRegionManager: onHeadsUpPinnedModeChanged
08-16 13:50:33.943   382   382 D StatusBar: disable<e i a s b h r c s > disable2<q i n >
```

logcat on SM-A1560 is so messy that I have no idea how to treat it. I'm sorry.

Just-Explode, 2024-08-16:
oh wait... I see that after Android 11 (API 30) you need to somehow grant the user ACCESS_FINE_LOCATION then ask them for ACCESS_BACKGROUND_LOCATION. (will this be persistence?) after reading [](https://developer.android.com/develop/sensors-and-location/location/permissions?hl=en#request-background-location) and the issue you mentioned.

Idk if I can help, but i'd like to try. But idk how's the progress now yet. (sorry) I think your can add two button in the Setting/PublicAPI option, one of them is "Get GPS permission", and the other "Allow running in the background" will be present or enable after clicking the above one? Since the user is already getting their hands on APIs, I think we can just let the user know the fact that it's Android forcing us to do so?
Sorry if you guys already implemented that.

Just-Explode, 2024-08-16:
lol idk why but I managed to learned how to do basic modification to this app with ChatGPT.
I did what I did to the app. Here's a screenshot (forgive me but i use Traditional chinese with meanless strings)
[image]

Welp, now I do have a checkbox called always allow location permission in setting ( i think it's caused by edit to AndroidManifest.xml?). But still this doesn't change anything.
I'm now thinking about it was either the way this app initiates, or how it stops recording from app or intent somehow stops the ACCESS_FINE_LOCATION for ACCESS_BACKGROUND_LOCATION? idk but i'll try

Just-Explode, 2024-08-16:
nvm i wanna do something else for now. reading java code is a huge pain to me even with ChatGPT

sgofferj, 2024-08-20:
I would think, it is unnecessary, because this issue popped up somewhen in Spring of this year. Probably some change in the code broke the functionality.

MaximusdaVII, 2024-09-18:
I think it has to do with a change in the Location permissions. I found an old apk of Opentracks v4.8.0 where the location permission looks like this:
[image]
On the Versions since about Spring this Year they look like this:
[image]
There is no way to allow Location Access all the Time.
I hope there is a way to bring this back.

abdoulayeYATERA, 2024-09-25:
Looks like a permission issue:

android.app.ForegroundServiceStartNotAllowedException: Service.startForeground() not allowed due to mAllowStartForeground false

~~~
Exception in thread "main": android.app.ForegroundServiceStartNotAllowedException: Service.startForeground() not allowed due to mAllowStartForeground false: service de.dennisguse.opentracks.debug/de.dennisguse.opentracks.services.TrackRecordingService
	at android.app.ForegroundServiceStartNotAllowedException$1.createFromParcel(ForegroundServiceStartNotAllowedException.java:54)
	at android.app.ForegroundServiceStartNotAllowedException$1.createFromParcel(ForegroundServiceStartNotAllowedException.java:50)
	at android.os.Parcel.readParcelableInternal(Parcel.java:5016)
	at android.os.Parcel.readParcelable(Parcel.java:4998)
	at android.os.Parcel.createExceptionOrNull(Parcel.java:3178)
	at android.os.Parcel.createException(Parcel.java:3167)
	at android.os.Parcel.readException(Parcel.java:3150)
	at android.os.Parcel.readException(Parcel.java:3092)
	at android.app.IActivityManager$Stub$Proxy.setServiceForeground(IActivityManager.java:6960)
	at android.app.Service.startForeground(Service.java:863)
	at androidx.core.app.ServiceCompat$Api34Impl.startForeground(ServiceCompat.java:241)
	at androidx.core.app.ServiceCompat.startForeground(ServiceCompat.java:172)
	at de.dennisguse.opentracks.services.TrackRecordingService.startSensors(TrackRecordingService.java:217)
	at de.dennisguse.opentracks.services.TrackRecordingService.startRecording(TrackRecordingService.java:195)
	at de.dennisguse.opentracks.services.TrackRecordingService.startNewTrack(TrackRecordingService.java:175)
	at de.dennisguse.opentracks.publicapi.StartRecording.execute(StartRecording.java:28)
	at de.dennisguse.opentracks.publicapi.AbstractAPIActivity.lambda$new$0(AbstractAPIActivity.java:22)
	at de.dennisguse.opentracks.publicapi.AbstractAPIActivity.$r8$lambda$yfwvwPfCJ0TKcPOvTGugdZ7vk5g(Unknown Source:0)
	at de.dennisguse.opentracks.publicapi.AbstractAPIActivity$$ExternalSyntheticLambda1.onConnected(D8$$SyntheticClass:0)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.lambda$execute$1(TrackRecordingServiceConnection.java:154)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection$$ExternalSyntheticLambda2.onConnected(D8$$SyntheticClass:0)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.setTrackRecordingService(TrackRecordingServiceConnection.java:135)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.-$$Nest$msetTrackRecordingService(Unknown Source:0)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection$1.onServiceConnected(TrackRecordingServiceConnection.java:62)
	at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:2206)
	at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:2239)
	at android.os.Handler.handleCallback(Handler.java:959)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loopOnce(Looper.java:232)
	at android.os.Looper.loop(Looper.java:317)
	at android.app.ActivityThread.main(ActivityThread.java:8592)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:580)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:878)

~~~

Just-Explode, 2024-09-25:
> There is no way to allow Location Access all the Time.
> I hope there is a way to bring this back.

I don't think there's a need for background permission. I read that app can have permission to do so if runs as foreground service(with a notification) from google's doc.
Correct me if I'm wrong, hopefully the devs knows what should happen

abdoulayeYATERA, 2024-09-25:
> I don't think there's a need for background permission. I read that app can have permission to do so if runs as foreground service(with a notification) from google's doc. Correct me if I'm wrong, [...]

Indeed, the service is trying to launch foreground notification while being launched from background (intent). That's the issue it seems.

abdoulayeYATERA, 2024-09-25:
@Just-Explode Here a fix

Just changed the AbtractApiActivity.java  on create fonction.

Replaced 
~~~
TrackRecordingServiceConnection.execute(this, serviceConnectedCallback);

~~~

By 
~~~
TrackRecordingServiceConnection.executeForeground(this, serviceConnectedCallback);

~~~

Not a all a stable fix ok.
No need to send you the apk, saw that you are able to compile already.
Just make that small change and install, should be good (I hope haha).

Just-Explode, 2024-09-25:
> Replaced 
> ~~~
> TrackRecordingServiceConnection.execute(this, serviceConnectedCallback);
> 
> ~~~
> 
> By 
> ~~~
> TrackRecordingServiceConnection.executeForeground(this, serviceConnectedCallback);
> 
> ~~~

OMG it's really glad to see a potential fix, I will definitely try!

commit 4a61c6ed (2024-12-05): Bugfix: PublicAPI StartRecording recorded no data.

Problem was that the TrackRecordingService was not started as a ForegroundService anymore.

Fixes #1904.
PR #2024 (2024-12-05): Bugfix: PublicAPI StartRecording recorded no data.
Problem was that the TrackRecordingService was not started as a ForegroundService anymore.

Fixes #1904.
commit b65230bc (2024-12-05): Bugfix: PublicAPI StartRecording recorded no data.

Problem was that the TrackRecordingService was not started as a ForegroundService anymore.

Fixes #1904.
closed 2024-12-05 by dennisguse