OpenTracksApp/OpenTracks #1905: Loss of GPS results in extreme distances

slackline, 2024-04-28:
## Describe the bug

I've noticed recently when at the end of a workout if I enter a house or building and lose GPS signal OpenTracks will guess at a location which results in ridiculous distances being recorded.

Obviously I'd be delighted to be able to get the reported statistics but know I'm just cheating myself!

I realise this may be an issue with GPS on the device but its new/recent behaviour that I've not seen before. Perhaps there is a way of ignoring GPS points if they fall outside a certain radius of the last X points?

## To Reproduce

1. Record a track.
2. Go into a building where GPS signal will be lost.
3. Wait.
4. See screenshots

### Screenshots

[image]
[image]

### Logcat

I've tried extracting a `logcat` but am unsure what to search for. :shrug: 

**Technical information**
 - Device: Motorola gt100 (XT2125-4)
 - OS: `LineageOS 21-20240311-NIGHTLY-nio`
 - OpenTracks version: `v4.12.2` (Version code : `5749`)
 - OpenTracks commit id (only for nightly builds): N/A

dennisguse, 2024-04-28:
Duplicate of https://github.com/OpenTracksApp/OpenTracks/issues/1887.
It is a bug in the idle handling.
#1887 also has some workarounds.
Fixed version will be released in some minutes.

closed 2024-04-28 by dennisguse

slackline, 2024-04-28:
Thanks @dennisguse I had a scan through issues but didn't make the connection.