OpenTracksApp/OpenTracks #1906: Data recorded but messed up due to single (maybe a few)  measurement error(s) ~ 11.000 km off

Humanimaliberation, 2024-04-29:
**Description** 

Track was recorded. Data seems mostly fine except one/some huge measurement error(s).

- Real distance ~34 km (as tracked in other App) 
- Shown distance >11.000 km, max. Speed >2.000 km/h
- Recorded time: 5h 9min although movement stopped after 4h XXmin  (forgot to stop tracking after finished tour)
- Speed / Distance Plot does not plot error correctly 
  - middle part over tousends of km: ~10 km/h linear going up, not >25.000 km/h as interval indicate
  - start of x-axis: seems like crouched valid data

Apart of the single (or few) wrong datapoint(s) the trackdata seems fine.

Indicators:

- Intervals seem realistic for first 34 km (realistic average speed shown)
- Speed / time plot seems fine (does not show the weird peak, y-axis adjusted to "real" max. speed)

Not sure if wrong datapoint(s) is/are at the total end of dataset or if there's valid data afterwards.
Cant scroll intervals manually that far, didn't check the exportable data yet.
(Did stop tracking for other app 7min earlier, also after some time not moving.)

Reproduction: 

 - Unknown (cause not identified, but other app showed valid data for that day) 

Fix proposal: 
Maybe integrate some plausibility check? If the distance from one datapoint to the last valid one is too huge, discard datapoint? 

First time that happened, otherwise App seems to be working quite accurate. 
(speed and distance measurements similar compared with other apps)

...

**Checklist**

- [ ] Consulted [https://dontkillmyapp.com/] (Did not check)
- [x] Power management: battery optimization is disabled for OpenTracks

**Technical information**

- Device: [Doogee S89]
- OS: [Android 12.0-20]
- OpenTracks version: [v.4.12.2]

**Images**

[image]
[image]
[image]
[image]

dennisguse, 2024-04-29:
Duplicate of #1898. 

Please upgrade to v4.12.3 (released yesterday). 

Sorry for the inconvenience.

closed 2024-04-29 by dennisguse

Humanimaliberation, 2024-04-29:
No worries and thx for the reply! 
And the app anyway.^^ 
Next I'll check open issues more firmly :sweat_smile: