OpenTracksApp/OpenTracks #1909: same Bewegungszeit and Gesamtzeit v4.12.2

Filefabrik, 2024-05-02:
**Describe the bug**
A clear and concise description of what the bug is.
"Bewegungszeit" and "Gesamtzeit" are the same values since the update to version 
**To Reproduce**

If applicable:
* add screenshots
* add logcat output

**Technical information**
[image]

 - Device: Samsung Tab s2
 - OS: android 10
 - OpenTracks version:  v4.12.2 5470

Filefabrik, 2024-05-02:
In addition: i had have started the recording but forgotten the tablet at home.
...later i was back and i see this screen. 
Hope that helps
[image]

dennisguse, 2024-05-02:
The first problem: looks like there is something broken. 

The 2nd one is fixed in v4.12.3.

Filefabrik, 2024-05-03:
Okay, thx 
then i will wait until  v4.12.3 is played in fdroid...

Grimtown, 2024-05-03:
Can confirm.

"total time" and "moving time" are the same, no matter how few or many breaks I take.

Nokia XR20
Android 13
OpenTracks v4.12.2

dennisguse, 2024-05-03:
Seems to be a duplicate of #1756. 
And yes something is broken or was never working with the new idle functionality.

closed 2024-05-03 by dennisguse