OpenTracksApp/OpenTracks #1912: Export Statistic Summary

G-i-o, 2024-05-09:
Hello.

**Describe the solution you'd like**
I would like to be able to export all recordings (or selected ones)  into a single CSV file containing the statistics for each track, such as DATE, ACTIVITY TYPE, DISTANCE, MOVING TIME....

Values should be total for each recording as shown on the dashboard as opposed to the currently available export which dumps all tracked point.

**Additional context**
This is an output sample of the export:
```
start_date, activity, distance, moving_time (day:hrs:min:sec)
2024-05-08_T18:30, walking, 10.5km, 00:02:30:15
2024-05-01_T19:00, walking, 11.02km, 00:02:50:45
```
......

Thank You very much for this awesome app. 🙏
P.S. Paid donor. Happy to make a further donation for this feature.

dennisguse, 2024-12-19:
@G-i-o what do you want to do with this data?
Is the CSV export for the tracks not sufficient?

G-i-o, 2024-12-19:
Hi @dennisguse .

I want to be able to import the data into my database and/or spreadsheet to be able to see for how long I carry each single activity or their summary during a period.

I do not know how to automatically convert the current exports (trackpoints) into actual data as shown on the OT dashboard (aka Km, hrs:mins ....).

I am aware the OT do offer some summary of the activities in itself, however, I need that data in my own database so that I can connect it and visualise it with the other activities as well as have more flexibility in general.