OpenTracksApp/OpenTracks #1925: Persistant Speed and Candence values

OMTDesign, 2024-06-07:
**Describe the bug**
I am using CYCPLUS C3 and S3 candence and speed sensors paired to Opentracks to record my cycling data. When I come to a complete stop while cycling the last recorded cadence and speed values are held, and both values does don't drop to zero. Since the speed is not reduced to zero, it prevents the app from going into the "idle" state. Because the app thinks I'm moving it records these incorrect values, which reduces my moving speed average.

**To Reproduce**

1. In settings, Sensors, under Cycling, add Cadence sensor and Distance and Speed Sensor.
2. Set wheel diameter to 2105mm
3. In Settings, User Interface, Customize your recording Layout, Add the Cadence data field.
4. Start a new recording and start riding. Come to a complete stop and see the Speed and Cadence values are not zero.
5. The "idle" state is not induced, so the app still thinks you are "moving" and records this data.

**Technical information**
 - Device:  Samsung Galaxy Note 9
 - OS: Android 10, OneUI 2.5
 - OpenTracks version: 4.12.4 version code 5788

**If applicable**
Attached is a video of this error. I am showing the outputs of Opentracks with the offical ANT+ Plugin Sampler. I was freewheeling (pedaling backwards) in the first half of the video for the cadence data, and I was pedaling forward for the speed data. There is a error in the first half of the recording where speed is reported while the wheel is stationary. This was likely due to signal interference as has solved itself.

https://github.com/OpenTracksApp/OpenTracks/assets/39666543/819bbb86-080f-4fda-b9e7-22fb110e6073

dennisguse, 2024-06-09:
Moving time is broken due to #1756. 

IIRC there is a 5s TTL for the sensor data. 
I need to check the computation for cadence if no new wheel revolution is reported. 
And you are right, it should drop back to 0.

OMTDesign, 2024-06-10:
At the time I did not know the delay for senor data was 5 seconds. From what I observed from the ANT+ Plugin Sampler was a 3 second delay from stopping pedaling to the value being updated / recorded as 0.  

The following is a graph for a bike ride I went on. When stopped at traffic lights, Opentracks holds the last cadence value, which results in a flat line. But for Speed it shows a short flat line, then a negative linear plot to the next value. 
However as shown in this video, the reported speed value is fixed and does not decrease as this graph indicates. I did manage to get it to report zero for speed, but that only happened when the sensor turned itself off due to inactivity for >3minutes.

[image]

https://github.com/OpenTracksApp/OpenTracks/assets/39666543/63c21c81-11b6-46cd-87f3-74bd0c7a9fc3

dennisguse, 2024-12-19:
@OMTDesign can you check again if this problem still occurs?

closed 2024-12-29 by dennisguse