OpenTracksApp/OpenTracks #1934: list not showing track being recorded

nutpantz, 2024-06-22:
**Describe the bug**
A track of being recorded but the list of tracks does not show the top one recoding

**To Reproduce**
Start a track. 
Go do something for 3 mins
Look at list of tracks.
None are listed as recording

If applicable:
* add screenshots
* add logcat output

**Technical information**
 Pixel 8 pro android 14
 - Google android 14 March 2024
[image]

The top track is actively recording

 - 4.2.14
 - OpenTracks commit id (only for nightly builds): [e.g. 4b333bb]

nutpantz, 2024-06-22:
Here are some Screen shots. One is after I stopped recording it still showed recording (you can see the button on the bottom red showing no recording). So there are more issues with displaying "recording"
[image]
[image]
[image]

pstorch, 2024-09-07:
I cannot reproduce this one as described, but in a different way.
- start a new recording
- stop the recording from the track details page
- the confirmation dialog is shown
- instead of finishing the track, hit the play button to resume
- recording is started again and the track list is show without any visible recording

PR #1983 (2024-09-08): Bugfix: Track list not updating recording status correctly.
One should react to underlying data changes if it changes, right?

Fixes #1934.
Fixes #1982.
commit d04e75c4 (2024-09-08): Bugfix: Track list not updating recording status correctly.

Fixes #1934.
Fixes #1982.
closed 2024-09-08 by dennisguse