OpenTracksApp/OpenTracks #1935: displaying elevation gain broken?

nutpantz, 2024-06-22:
**Describe the bug**
Showing elevation gain is broken.

**To Reproduce**
Start track
Go up and down something 100 meters
Look for elevation gain

If applicable:
* add screenshots
* add logcat output

**Technical informatig
Pixel 8 pro Google android 14 March update
 - v3.2.14
 - OpenTracks commit id (only for nightly builds): [e.g. 4b333bb]
[image]
[image]

You can see elevation has been recorded

nutpantz, 2024-06-22:
Also elevation gain is not in stats like it used to be after a track is stopped

[image]

nutpantz, 2024-06-22:
Just because I should be complete in my testing and reporting .
Also not showing elevation in intervals page
[image]
![Screenshot_20240622-151904](https://github.com/OpenTracksApp/OpenTracks/assets/12160507/e759cffb-e973-4a0a-8

nutpantz, 2024-06-22:
Lots of elevation gain and loss
[image]

paulle, 2024-06-28:
I can confirm this. Here te same.
Xperia 5 V, Android 14

dennisguse, 2024-07-11:
Is the barometer configured in the settings?

paulle, 2024-07-11:
No, will try it next time.
Thank you

-------- Ursprüngliche Nachricht --------

paulle, 2024-07-11:
It works

-------- Ursprüngliche Nachricht --------

closed 2024-07-11 by dennisguse

nutpantz, 2024-07-12:
What is wrong with the gps only altitude?

dennisguse, 2024-07-12:
@nutpantz GPS altitude: absolute measure (with quite some noise). 
Barometer altitude: relative measure (with drift due to weather changes).

Nothing wrong: just different data sources that can be used for different things.