OpenTracksApp/OpenTracks #1950: Determine the wheel circumference of a bicycle speed sensor from GPS

dr-m, 2024-07-05:
Related to #1949 it would be nice if OpenTracks could determine the bicycle wheel circumference on its own, so that I did not have to remember and type it when switching bicycles.

There could be a menu item that is available during a recording or , to calibrate the wheel size. The user would have to activate it manually. The result would apply to the entire current recording and any subsequent recordings.

Alternatively, near the start of a GPS based recording, OpenTracks can detect when I have been riding straight on a level surface for a few tens or hundreds of meters. If the calculated wheel circumference differs from the configured one, it would display a message with the options "adjust" or "ignore". I think that this would be problematic in various respects: What exactly is "near the start"? What if one starts at a steep hill or in an area of bad GPS reception? What if the display was already turned off?

Random90, 2024-07-09:
Seems like a big feature for an edge case, better proposal for now is to add an ability to save different bikes configurations and allow to change them with 2 taps on the main screen. Not only the circumference, but maybe also specific sensors.

dennisguse, 2024-09-08:
@Random90 you are right - this could be handled by implementing proper profiles.
Let's track this in #1954.

closed 2024-09-08 by dennisguse