OpenTracksApp/OpenTracks #1957: RFE - allow adding markers after finishing track

dhlocker, 2024-07-12:
**Is your feature request related to a problem? Please describe.**
This is a straight up request for enhancement.
I have no current troubles, but would like to have the ability to insert a marker after finishing a track.

**Describe the solution you'd like**
Scenario:
- Moving through space;
- Have to reverse because of blockage (DON'T want to stop to add marker because of losing time);
- Continue to end and finish track;
- Now would be a good time to go back and add a marker and annotation at the location of the blockage;
- Oops - can't mark the track at the blockage

**Describe the solution you'd like**
Ideal solution would be:
- Open finished track
- View map
- Click three dots
- Select "Edit Track"
- Tap on track at significant location
- Choose "Add Marker"
- Edit marker information
- Click "Done"

**Describe alternatives you've considered**
I did try "continue track" after having finishing it, but it appears markers can only be added at the current location/time.

**Additional context**

pstorch, 2024-07-12:
With "view map" and "Tap on track at significant location" you are outside of OpenTracks. The tracks are readonly at the moment for other apps. To make this possible we would need a write API back to OT. I don't know if this should be allowed.

dhlocker, 2024-07-12:
I understand that OSM Dashboard provides the Open Tracks map display functions. I am not familiar with the OT API, so my suggestion was clearly very naive. I can understand how adding a publicly accessible call into OpenTracks to modify a track could open many security holes.

Perhaps another means to accomplish a similar effect: allow a user to select a point on the "By distance" or "By time" display to insert a mark. (The user would have to then look at the map to see if it is "close enough" to the desired location and whether to edit, delete, or create a new mark.)

Just brainstorming at the kbd, now. If this is all too much pipe dream, please close.

pstorch, 2024-07-13:
Maybe the OSMDashboard can send an "addMarker" Intent to OT with the tapped coordinates.
Might not be that difficult after all.

dennisguse, 2024-07-14:
The MarkerEditActivity needs be called for this. 
One nitpick: Markers also should have a time (e.g. when they were created while tracking). 
Otherwise, they will be shown in the beginning in the time diagram.

pstorch, 2024-07-14:
Looks like an interesting UseCase to me. I'll give it a try.

pstorch, 2024-07-15:
I got a working solution:

- https://github.com/OpenTracksApp/OpenTracks/pull/1960
- https://github.com/OpenTracksApp/OSMDashboard/pull/388

But I'm not sure, if this is ok on OpenTracks side. The EditMarkerActivity is using the TrackRecordingService and Manager which are dependent on a running recording. I had to "punch a hole" in the insertMarker code path to be able to pass the given location to a finished recording. Why does the Marker need the TrackStatistics (length and duration)?

@dennisguse what do you think?

dennisguse, 2024-07-15:
Without reading the code:
* You can change MarkerEditActivity to get lat/lng + time and distance via Intent. Just move the TrackRecordingService to wherever it got called from so far
* And use a different activity (like StartRecording) to trigger MarkerEditActivity

closed 2024-07-30 by pstorch

pstorch, 2024-07-30:
@dhlocker if you like, you can test this functionality in the nightly builds.

dhlocker, 2024-07-30:
> @dhlocker if you like, you can test this functionality in the nightly builds.

I'm unsure how to fetch a nightly build. It's not visible in F-droid and I don't see a nightly build in github (I'm VERY unfamiliar with github). Clues are welcome - I'd love to try this enhancement.

pstorch, 2024-07-31:
Yes, they can coexist.

dhlocker, 2024-08-20:
My apologies for the long delay. That worked beautifully!

commit 389c4bed (2024-07-30): fix #1957 open MarkerEditActivity to receive external marker (#1960)

* fix #1957 add CreateMarkerActivity to receive external marker