OpenTracksApp/OpenTracks #1958: Activity list titles are being truncated

furo321, 2024-07-12:
**Describe the bug**
Activity list titles are being truncated, despite there being plenty of space. See screenshots. Happens more commonly with longer titles, sometimes multiple activities or just one. It usually changes which ones are being truncated when going in and out of an activity.

[image]

[image]

**Technical information**
 - Device: OnePlus Nord CE
 - OS: LineageOS 18.1 (Android 11)
 - OpenTracks version: v4.12.4 (F-Droid)

Pulau-Komodo, 2024-07-13:
The number of characters before truncation seems random for each item, and it seems to re-randomize every time I go into an activity and back to the list.

dennisguse, 2024-07-22:
I guess it depends on the list. 
The UI framework does not recompute the required size of the names and then it just truncates them.

commit ca70b70b (2024-09-07): Bugfix: Track list should use available space for track names.

Fixes: #1958.
closed 2024-09-07 by dennisguse