OpenTracksApp/OpenTracks #1959: Add option to display all or selected tracks on the single map

czarnyckm, 2024-07-14:
It will be good to have an option to display all or selected tracks on a single map

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

pstorch, 2024-07-14:
You can already do that.

czarnyckm, 2024-07-18:
Already implemented. Thanks

czarnyckm, 2024-07-18:
Already implemented

closed 2024-07-18 by czarnyckm