OpenTracksApp/OpenTracks #1963: UI/UX issue: multiple date formats make figuring out when something happened confusing

SamWhited, 2024-07-24:
**Is your feature request related to a problem? Please describe.**

When viewing the list of recorded tracks, multiple date formats are used in the same place, making it confusing to figure out when things are happening. The first few tracks use a relative date format ("Today", "Yesterday"), then the next few use a day of the week ("Friday", "Saturday") presumably meaning the previous week days, though it's very unclear to me what the date is, then beyond that it actually uses the correct date (so these are the only ones where I always know what day a thing happened on without having to stop and think).

**Describe the solution you'd like**

All track recordings should just use the actual date.

**Additional context**

![A screenshot from OpenTracks, a track recording app for Android. Several bike rides are shown. The first two in the date field say "Today" and "Yesterday", the second two say "Monday" and "Friday" and the last two show actual dates.](https://github.com/user-attachments/assets/ebf0fabc-ace6-44f6-bb11-47fdefcdf8a4)

dennisguse, 2024-07-26:
@SamWhited just open a PR for this.

SamWhited, 2024-07-26:
I don't really do Android stuff anymore or have any sort of dev setup to do it, sorry. Not demanding a quick fix or anything, I just wanted to file/document the issue. Sorry I can't be the one to fix it.

wasif1, 2024-10-20:
you want actual date for all ? Please tell me i will do it. Assign it to me.