OpenTracksApp/OpenTracks #1965: Trekking mode (energy-efficient profile)

ksz16, 2024-07-29:
Hi, new user here. 
So far I have been using [AAT](https://github.com/bailuk/AAT), but I decided to give OpenTracks a chance and so far I have a very positive feeling.

A feature that would definitely be worth adding are profiles @https://github.com/OpenTracksApp/OpenTracks/issues/1954
In addition to the standard profiles based on activity type, it would be nice to have a special profile focused on energy conservation. It would work great in use scenarios such as trekking (hiking or biking), when access to energy sources is difficult and run time per charge is crucial.

For now, I've started testing OpenTracks and I can say that it's lightweight and quite battery-friendly (compared to [Osmand](https://github.com/osmandapp/Osmand), which consumes a lot of resources). But there are applications that are even more energy efficient - I'm thinking of [ulogger](https://github.com/bfabiszewski/ulogger-android). I wonder if it would be possible to create such a profile in OpenTracks (once this feature is implemented, of course), which would be focused on extending battery life at the expense of recording detail how does [ulogger ](https://github.com/bfabiszewski/ulogger-android )do.

In such a profile, it would not be possible to use external sensors or check the current position on a map. All it would do is work in the background and record track. The logging intervals should, of course, be adjusted accordingly to consume as little power as possible. This would come at the expense of accuracy, but that's the price.

Currently I use [ulogger ](https://github.com/bfabiszewski/ulogger-android )in such situations, but it would be nice to have such a function in OpenTracks and thus have everything in one place.

dennisguse, 2024-08-04:
Interesting idea!
One question: how does OpenTracks power consumption compare to ulogger?

ksz16, 2024-08-05:
First of all I wanted to thank you for your work and creating such a cool app. Thank you also for your interest in the trekking mode topic.

Unfortunately, I do not have accurate measurements of energy consumption. But I will use RAM consumption. Tested on OnePlus 6 (LineageOS 20).

OpenTracks immediately after starting consumes about 44 MB. 10 minutes later, the consumption increases to about 56 MB. After 45 minutes it is about 60 MB. If we check the position on the map (via OT Dashboard API) the consumption increases to about 70 MB. From my observations, using the map while writing the path increases RAM consumption up to about 90 MB.

In contrast, ulogger immediately after starting consumes 20-22 MB. This consumption is more or less constant throughout the session (even if it lasts 8 or 10 hours) and oscillates between 20-25 MB.

I haven't yet had a chance to test OpenTracks under such conditions (I mean continuous use for 8 - 12 hours), but I have used ulogger in this way and I can say that its impact on battery consumption is really small.

And finally, a note: of course, the track saved with ulogger is not as accurate as in OpenTracks (fewer logging points), but sometimes it is better to have a track less accurate than not to have one at all.

dennisguse, 2024-09-08:
Memory consumption is not really a good indicator for power consumption.
And hopefully, we have memory leak in the code though.

For actually improving power consumption:
I guess, what we would need to implement is a proper profile support (aka applying different settings).
Let's track this in  #1954.

closed 2024-09-08 by dennisguse