OpenTracksApp/OpenTracks #1970: Selecting application language from a list rather than using Android system language

Patriccollu, 2024-08-15:
Hello,

As [**Corsican** translator](https://github.com/Patriccollu/Lingua_Corsa-Infurmatica#readme), I would be very happy to contribute to this very interesting project by adding this brand new language to OpenTracks.

But before doing this, I need to know whether there is a _language switcher_ within the application. As Corsican is not yet recognized as system language on Android, the only way to use an application in Corsican is to manually select it from a list of available languages.

Do you have such language switcher option in OpenTracks?

Best regards,
Patriccollu.

dennisguse, 2024-08-16:
OpenTracks only uses the Android locales. 
If you want to extend that, feel free to.
Adding a locale chooser into OpenTracks is definitely possible - not sure how much effort. 
Feel free to extend it :)

Patriccollu, 2024-08-16:
@dennisguse Thanks for your answer.

I would be very happy to help but I need some clarification.

When searching previous issue/PR related to UI and languages, we can find the following:
- [Android system per-app language preferences](https://github.com/OpenTracksApp/OpenTracks/pull/1468)
- [Upgrade to AGP 8.1.0-alpha](https://github.com/OpenTracksApp/OpenTracks/pull/1521)
- [UI shows incorrect language](https://github.com/OpenTracksApp/OpenTracks/issues/1527)
- [Build: remove resConfigs limiting included languages](https://github.com/OpenTracksApp/OpenTracks/pull/1540)

PR #1468 was using `src/main/res/xml/locales_config.xml` and `resConfigs` section in `build.gradle`, then PR #1521 deleted `locales_config.xml` file and PR #1540 removed `resConfigs` section`.

Does that mean that selecting the UI language from a list was a functionality which was existing before but disappeared in April/May 2023?

Please can you clarify?

Patriccollu.

dennisguse, 2024-08-17:
No, all of this was manual configuration required back then. 
This configuration is now generated by the Android Gradle Plugin during the build process. 
To my understanding, it checks which languages are there (strings.xml files) and then creates the configuration.

Patriccollu, 2024-08-17:
Hi Dennis,

Thanks for having clarified.

Unfortunately my development's knowledges do not allow me to implement a locale chooser!

But if this feature is developed by someone else, I will be more than happy to contribute as Corsican translator!

So I'm closing this issue.

Best regards,
Patriccollu.

Patriccollu, 2024-08-17:
Closing as not planned.

dennisguse, 2024-08-17:
It feels more like configuring this properly.
Just check the documentation and play around with it: https://developer.android.com/guide/topics/resources/app-languages

Patriccollu, 2024-08-17:
I appreciate your help but I'm really sorry: I'm just ;-) a translator!
In the past I was able to add the Corsican language code in an existing list of languages and a couple of similar 'simple updates' for several applications but that's all.

Nevertheless, I started to read the _Android Developer's Guide_ using the link you provided. If I understand:
- you are using AGP 8.1 (or more) so `LocaleConfig` file is generated and does no longer need to be maintained manually
- and `build.gradle` file contains `android { androidResources { generateLocaleConfig true } }`
- when Corsican language will be created and populated in Weblate, a `\res\values-co\strings.xml` will be generated, and used
- my smartphone still using Android version 11, the guide recommends using the AndroidX support library when implementing an in-app language picker
- `appcompat:1.7.0` is implemented but `setApplicationLocales()` and `getApplicationLocales()` methods look to be reserved for Android 13 and higher
- `OpenTracks/src/main/AndroidManifest.xml` needs to be updated using instructions in [Support Android 12 and lower](https://developer.android.com/guide/topics/resources/app-languages#android12-impl) section
- ... but that 's probably not enough!

dennisguse, 2024-08-18:
I did some tinkering around and I got something working.
Sadly, it will not be only configuration as the per-app language picker is not picking up `co` as language code from the locale_config.xml (also generating this file automatically doesn't work).
However, the locale can be set programmatically:
```
        var appLocale =
                LocaleListCompat.forLanguageTags("co");
        AppCompatDelegate.setApplicationLocales(appLocale);
```
Funnily, this will show 'Corsian' in Android's per app setting as selected locale ;)

So, you can just create `\res\values-co\strings.xml`via Weblate and in OpenTracks' settings, we add an override system locale configuration setting (until Android get's fixed).
Have fun translating :D

Patriccollu, 2024-08-18:
@dennisguse Great!
I'm starting translating and you will hear from me in a couple of weeks!

Patriccollu, 2024-08-20:
@dennisguse I just completed the translation of `strings.xml` file on [Weblate](https://hosted.weblate.org/projects/opentracks/-/co/) and opened new issue #1974  for that purpose.

Thanks again for your support and research on allowing Corsican to be selected from a list of available languages.

I'm closing the current issue.

closed 2024-08-20 by Patriccollu