OpenTracksApp/OpenTracks #1971: Unable to stop activity when using gestures system navigation

JanezCZ, 2024-08-16:
**Describe the bug**
When Android system is set to gestures navigation and help is hidden, it is almost impossible to stop the activity.
It seems like system is taking long press as start of a gesture and therefore does not forward press to an app.

**To Reproduce**
1. Set system to gestures navigation
2. Hide navigation hint
3. Start activity
4. Try to hold stop button

**Technical information**
 - Device: Fairphone 3
 - OS: /e/OS 2.2-t-20240715417772-stable-FP3
 - OpenTracks version: 4.13.0 (code 5837)

**Screenshots:**
App with navigation hint enabled. Works properly
[image]

App with navigation hint disabled. Hold to stop not working:
[image]

dennisguse, 2024-08-16:
Can you create a pull request for this?

My phone lacks that option to hide the gesture hint.

dennisguse, 2024-09-08:
Duplicate of #1893.

closed 2024-09-08 by dennisguse