OpenTracksApp/OpenTracks #1972: Option to automatically upload data to OpenStreetMap

vfosnar, 2024-08-16:
**Is your feature request related to a problem? Please describe.**
I'd love to be able to automatically contribute my tracks to OSM.

**Describe the solution you'd like**
Settings option to automatically send recorded tracks to OSM. The track beginning and end could/should be removed by a given radius to preserve privacy. There should be an option to upload anonymously (probably enabled by default).

**Describe alternatives you've considered**
Manually exporting and stripping tracks and then uploading them to OSM.

vfosnar, 2024-08-17:
https://github.com/OpenTracksApp/OpenTracks/issues/769#issuecomment-840732074

dennisguse, 2024-09-08:
@vfosnar if you want to work on this, please re-open.
Some concept without adding network permissions into OpenTracks would be nice though.
Just from the top of my head: after finishing a recording send an Intent, so that another app can handle the upload.

closed 2024-09-08 by dennisguse