OpenTracksApp/OpenTracks #1974: Corsican localization is now ready on Weblate

Patriccollu, 2024-08-20:
Hello,

I just completed the **Corsican localization** of `opentracks-opentracks-src-main-res-values-strings-xml-main` file on [Weblate](https://hosted.weblate.org/projects/opentracks/-/co/).

Please note that two other files - _App store metadata_ and _glossary_ do not contain any string to translate.

Best regards,
Patriccollu.

dennisguse, 2024-08-27:
Merged localization, but we still need a language switcher as Android's built-in doesn't pick up Corsican.
See: https://github.com/OpenTracksApp/OpenTracks/issues/1970#issuecomment-2295355736

Actually, this might even be considered a bug in Android.

Patriccollu, 2024-08-28:
Hope that you will be able to circumvent this problem by setting `co` locale programmatically!

Patriccollu, 2024-10-06:
Hi @dennisguse,

Do yoy have some news on this enhancement?

Best regards,
Patriccollu.

dennisguse, 2024-10-06:
Sadly no. 
Times are quite busy. 
I should have time at the end of the year. 
My plan was to use the approach taken by K9-mail and just copy their implementation - what could go wrong, right?

Patriccollu, 2024-10-06:
I can confirm that Corsican language can be selected on K9-Mail app.

pstorch, 2024-11-13:
Just stumbled upon this: https://apt.izzysoft.de/fdroid/index/apk/io.nightdavisao.multilocale

Would it help to make Corsican locale selectable? Haven't had time to test myself.

dennisguse, 2024-11-14:
@pstorch this tool needs root access.

pstorch, 2024-11-15:
@dennisguse there is also an option to grant permission via ADB. I want to try that.

pstorch, 2024-11-15:
I could give the app permission via adb, but it doesn't offer Corsican locale either.

dennisguse, 2024-12-01:
@Patriccollu feature is available as of v4.15.0 (will take some days for F-Droid).
However, only for Android 14 (SDK34+) or later - the commit messages says SDK33+, but that is a mistake on my side.
I couldn't figure out a way to get the list of available locales/languauges prior Android versions and I didn't want to manually set those somewhere.
Hope this works for you - if not we can keep digging :D

Patriccollu, 2024-12-06:
@dennisguse
I need to find an Android 14 device to be able to test as my current phone is only 11 :-(

Thank you for your work on this issue. I will keep you informed.

dennisguse, 2024-12-08:
@Patriccollu The reason for this is an API that lists the available locales that were bundled with the application.
Sadly, this API is only available on Android 14 or later.
Otherwise, we would manually need to maintain the lists of locales in the source code in addition to the actually translations from Weblate.

I also checked for alternatives (like `LocaleManagerCompat.getApplicationLocales()`), but those don't provide not all locales.

Patriccollu, 2024-12-08:
@dennisguse And what about the option to use the approach taken by K9-mail?

[image]

dennisguse, 2024-12-10:
Let's see...

dennisguse, 2024-12-11:
@Patriccollu should work now..
Implementation is brutally ugly, but hey it works and the locales will be updated automatically.
We are not a beauty contest, right? ¯\_(ツ)_/¯

closed 2024-12-11 by dennisguse

Patriccollu, 2024-12-12:
Hi @dennisguse ,

It's just... perfect! ;-)

Thanks a lot,
Patriccollu.
[image]

commit 4262e605 (2024-11-30): Settings: add custom language/locale chooser (SDK33+).
Corsican was not available using Android's per-app-language-chooser.

Fixes #1974.
PR #2021 (2024-11-30): Settings: add custom language/locale chooser (SDK33+).
Corsican was not available using Android's per-app-language-chooser.

Fixes #1974.

# Thanks for your contribution.

## PLEASE REMOVE
To support us in providing a nice (and fast) open-source experience:
1. Verify that the tests are passing
2. Check that the code is properly formatted (using AndroidStudio's autoformatter)
3. Provide write access to the [branch](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/allowing-changes-to-a-pull-request-branch-created-from-a-fork)
4. If the PR is not ready for review, ple [... cut]
commit 4313b51e (2024-11-30): Settings: add custom language/locale chooser (SDK33+).
Corsican was not available using Android's per-app-language-chooser.

Fixes #1974.