OpenTracksApp/OpenTracks #1995: Speed and cadence sensors display values >0 when idle detection detected a stop

lsuhrbier, 2024-10-10:
After having stopped the idle detection, non-zero values are displayed from speed and cadence sensors. It appears, the last measured values are shown and not reset to zero or not read again from the sensor after idle detection detected a stop.

dennisguse, 2024-10-14:
Code looks like this is a valid complaint.
Relevant code:
* TrackPointCreator.reset() [called by idle]
* SensorDataSet.reset()

Later on used by:
* TrackRecordingManager.getDataForUI()

Culprit may be children of Aggregator and the implementation of Aggregator.reset().
If I understood the code (I wrote), only accumulated values (like distance) are reset, but not the actual measurements (like speed, cadence, power).
So, the problem seems to be happening during the idle situation.

dennisguse, 2024-10-14:
PS/ you may be idle while a cadence sensor reports values - HR will do this as well :D

dennisguse, 2024-12-05:
Likely related: #1925

dennisguse, 2025-01-20:
The fix is broken... See #2043.

PR #2029 (2024-12-17): Bugfix: on idle detection reset movement-based sensor data.
When becoming idle, reset the sensor data, so it is only stored once.
If after becoming idle, the sensors report new data that's fine.

Fixes #1995.
commit fd9646bb (2024-12-17): Bugfix: on idle detection reset movement-based sensor data.

Fixes #1995.