OpenTracksApp/OpenTracks #1996: Can't add a photo to a track

k0styan3, 2024-10-11:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: [e.g. Nexus 7 (2013)]
 - OS: [e.g. LineageOS 15.1]
 - OpenTracks version: [e.g. v3.2.4]
 - OpenTracks commit id (only for nightly builds): [e.g. 4b333bb]
It is impossible to add a photo to a track - when trying to add a marker with a photo, there is simply no such option. In the list of options there is no option to add a photo. And there is no other way to add a photo Just no. I apologize in advance for my bad English - I am writing through a translator. My phone is android 13, stock firmware, no root. The program is installed from f-droid.

iiipt, 2024-10-12:
I have the same problem
Device: Nokia 3.1 +   (2018)
system: android one (android 10)

There just isn't the button to add image to marker

iiipt, 2024-10-12:
The osm dashboard works fine, when I click on marker it show the marker in the main app.

k0styan3, 2024-10-12:
> The osm dashboard works fine, when I click on marker it show the marker in the main app.

I don't understand where this is...can you attach a couple of screenshots?

iiipt, 2024-10-12:
[image]

You click on the red marker in the map and it will open the marker in the main app (this is my last marker where I was able to upload image, its from august)

[image]

k0styan3, 2024-10-12:
How do you even get from the tracker to the Axis Dashboard? When I click on
a track and then on the map icon at the top, I get a suggestion to open one
of two other applications, but it's  There is no OSM dashboard there!
С уважением, Константин С

iiipt, 2024-10-12:
You need to install it
here link: https://f-droid.org/packages/de.storchp.opentracks.osmplugin/

commit e5a60a81 (2024-10-12): Bugfix: UI for adding photos to marker was invisible.
Introduced in 00258fd93e6433338d5628a35656fe874e90ad1d

Fixes #1996.
closed 2024-10-12 by dennisguse