OpenTracksApp/OpenTracks #1999: gain/loss values seems not be reliable

ironspock, 2024-10-14:
recording multiple tracks with different elevation in "lower mountains" (max. elevation approx. 500 m)
some examples (in most cases the track were basically going up and then down):
gain / loss
1254 m / 1284m 	by max. elevation of approx. 480m (track starts at approx. 150m) 
129 m / 132 m	difference between lowest and highest elevation approx. 30m
1062 m/ 1062 m	by max. Elevation of approx. 500m (track starts at approx. 300m; lowest point at approx. 200m)
1164 m / 1164 m     by max. elevation of approx. 300 m (tracks starts at approx. 100m)

compared to "original" gain/loss values (e.g. from downloaded tracks) the recorded gain/loss values are approx. 3-4 times higher 

checking device based on https://github.com/OpenTracksApp/OpenTracks/blob/main/README_TESTED_SENSORS.md#tested-barometric-sensor-smartphones all 4 tests passed

**Technical information**
 - Device: Gigaset GX6
 - OS: Android 13
 - OpenTracks version: v.14.0 5889 F-Droid

dennisguse, 2024-10-14:
OpenTracks records each change that is +/- 3m.
So your phone's barometer reported this as a change.

The barometer is often also affected by how the device carried (like in a pocket or backpack or even using the case).
Difficult science :)

ironspock, 2024-10-16:
Hi Dennis,
it's really difficult science... :)
Based on your feedback I've run some "tests". Keeping the smartphone in hand loss/gain values are calculated as expected, carrying it in a pocket (e.g walking on same level a certain time/distance) you'll get really random loss/gain values.
So basically I'm aware of the "side effects" now.
Thanks for your information!
Regards ironspock

closed 2024-10-16 by ironspock