OpenTracksApp/OpenTracks #2003: Having access to smartphone altimeter, and being able to calibrate on top of it.

irrelevantspechack, 2024-10-16:
**Is your feature request related to a problem? Please describe.**
I'm frustrated when I see the point of opentracks and no elevation, because elevation require a bluetooth device that calculate it.

**Describe the solution you'd like**
I want opentracks, if the smartphone have one, to use the smartphone altimeter (as I understood it's linked to the smartphone barometer), and in addition, because I heard report of problems using this one, being able to correct the elevation on top (some users on others apps report that the altimeter indicate 12 m at sea level, and it's apparently a known problem that is linked to something else I didn't really understand). 

**Describe alternatives you've considered**
We still can export and use a software/app that will use the known elevation that are on the maps I guess. But the point is to have everything directly on opentracks. 

**Additional context**
none.

dennisguse, 2024-10-17:
1. Internal barometer is supported and should be selectable in the same menu as a Bluetooth barometer.

2. A barometer is used to compute gain/loss and not altitude. It works nicely for relative changes but not for an absolute value aka altitude. 

3. Altitude is derived from GPS and shown as EGM2008. That should be sufficiently enough. 

4. Air pressure to elevation is not fixed: it changes during the day (also due to weather changes). This also makes it tricky to use for absolute altitude. 
The effect of calibration of a barometer (air pressure) for gain/loss should be that big.
And it is very complicated to implement this correctly.

irrelevantspechack, 2024-10-17:
Ok, I would like the elevation done + / - in the stats directly. The problem is that elevation is in "by distance", but in "Intervals", it shows nothing, just distance, speed, and "elevation - m".

Idea is to have it in the stats everywhere. thanks for helping

dennisguse, 2024-10-18:
@irrelevantspechack this intervals screen also shows gain/loss (again barometer measured).
Thus, it will be "- m" if no barometer was used.
I will change the caption to gain/loss to make this clear.

commit a015b87c (2024-11-30): UI: interval list uses gain/loss as caption (instead of elevation).

Fixes #2003.
closed 2024-11-30 by dennisguse