OpenTracksApp/OpenTracks #2006: Can't open kmz in google earth -show nothing-

irrelevantspechack, 2024-10-17:
**Describe the bug**
When I export the kmz and open it with google earth, it doesn't display any route. Just a little white box with the name of the file when clicking on the left on the file, this little white box appear with "premiere" under the name of the file.

**To Reproduce**
1. Go to  opentracks, record a gps track, stop. 
2. Click on share and open with google earth/download it and open it with google earth
3. 
4. See error : Only "premiere" with white box appear

If applicable:
* add screenshots
* add logcat output

**Technical information**
[image]

 - Device: Recent I5
 - OS: Windows 11/Android
 - OpenTracks version:  last to date
 - OpenTracks commit id (only for nightly builds): [e.g. 4b333bb]

pstorch, 2024-10-18:
OpenTracks exports KML in Version 2.3, Google Earth only supports 2.2.

irrelevantspechack, 2024-10-18:
thnaks for the feedback, is there a way ton convert it back to kml 2.2?

thanks

pstorch, 2024-10-18:
@irrelevantspechack sorry, I don't know any converter from kml 2.3 to 2.2

closed 2024-10-18 by dennisguse