OpenTracksApp/OpenTracks #2008: App interface is off-screen

agichim, 2024-10-24:
**Describe the bug**
The top part of the app is "off-screen" in the sense that's it's shown under the camera and status bar icon, so they overlap.

Not important, but also the record button, is off the background of the bottom bar.

**To Reproduce**
1. Open the app.

If applicable:
* add screenshots

https://github.com/user-attachments/assets/1506668b-d0b3-45c1-955f-a62c1156f729

**Technical information**
 - Device: Google Pixel 7
 - OS: GrapheneOS 2024102100
 - OpenTracks version: 4.14.0

dennisguse, 2024-10-24:
This is due to Android 15 / TargetSDK 35. 
The upgrade in OpenTracks is not completely implemented: and the UI change is the one thing missing.
Kind of annoying to do in old code bases...

OpenTracks should still be functional. 
Is it?

For more info see #1955.

Options:
* Wait until it gets fixed (I might have time in December)
* Downgrade to a previous version of OpenTracks
* Downgrade to Android 14 ;)
* Fix it yourself...

jmizell, 2024-11-05:
Confirming v4.13.1 renders correctly on Android 15. Downgrading is atleast a reasonable workaround till the app can be updated.

PR #2017 (2024-11-19): Android15#2008
Fixes #2008.
Fixes #1955.
closed 2024-11-19 by dennisguse