OpenTracksApp/OpenTracks #2016: Make Developer Built .apk's Available on Github?

DiagonalArg, 2024-11-13:
I wonder if the team might consider making .apk's available here on github.  A number of us run GrapheneOS with [Obtainium](https://github.com/ImranR98/Obtainium).  That allows us to use developer signed packages directly from github or other developer websites, which is preferable to those signed by f-droid.

DiagonalArg, 2024-11-22:
Ah, I missed that.

Play Store has links to google libraries.  That would obviate the point of GOS.  I'll wait.  Thanks.

dennisguse, 2024-12-24:
Reproducible builds may work.
For now the PlayStore release is available via Github: https://github.com/OpenTracksApp/OpenTracks/releases/tag/v4.17.1

If reproducible builds work, I will publish the release to Github as well.

closed 2024-12-24 by dennisguse

dennisguse, 2024-12-24:
Anyhow, I would recommend fetching the reproducible build from Fdroid (after #418 is finished). 
Will be some days later, but it will be verified.

DiagonalArg, 2025-02-22:
@dennisguse / Just bumping this.  Any chance you might post those reproducible builds here?

dennisguse, 2025-02-22:
The builds are attached to the release on GitHub now. 
However, they should only be considered reproducible if the same APK appears on Fdroid.

DiagonalArg, 2025-02-23:
> Reproducible builds may work. For now the PlayStore release is available via Github: https://github.com/OpenTracksApp/OpenTracks/releases/tag/v4.17.1
> 
> If reproducible builds work, I will publish the release to Github as well.

I see the playstore release, but not a FOSS release as you had suggested you'd be willing to publish.  Would that be possible?

dennisguse, 2025-02-23:
The reproducible build variant has the suffix playstore as package name - as this was already built and signed by me. 
Reusing it was the easiest approach and now people can get the variant via Play Store or Fdroid. It does not matter :)

Changing the package name would have added more confusion. 

https://github.com/OpenTracksApp/OpenTracks/issues/418#issuecomment-2508970266

DiagonalArg, 2025-02-23:
Alright.  I see.  I'd say that's a bit confusing, as I would have called the new one ".fdroidrepro" or the like; but I can make do.  I wonder if you want to add a comment to your github releases page to the effect that .playstore is actually the reproducible fdroid release?  Otherwise we might think it contains the google play core library.

DiagonalArg, 2025-03-03:
Anonaddy uses the term "gplayless" to indicate it's *not* a version with google play libraries.  See eg. [here](https://gitlab.com/Stjin/anonaddy-android/-/releases).

ticklemyIP, 2025-03-07:
Alternatively devs use the 'FOSS' suffix.