OpenTracksApp/OpenTracks #2020: Need details to translate string 'voiceAnnounceLapPower'

Patriccollu, 2024-11-26:
Hello,

I'm Corsican translator for OpenTracks and I left untranslated this string `voiceAnnounceLapPower` so far. Please can you provide some details explaining how it is used in the application in order to provide an accurate translation.

I left a [comment on last August about this on Weblate](https://hosted.weblate.org/translate/opentracks/opentracks-src-main-res-values-strings-xml-main/co/?q=state%3A%3Ctranslated&offset=2#comments) but didn't receive any answer so far.

Thanks,
Patriccollu.

dennisguse, 2024-11-30:
It is about if for voice announcement, the power information (if a sensor is present) should be spoken.

PS/ I don't get notifications from Weblate (except something technically breaks), so asking here may be faster.

closed 2024-11-30 by dennisguse

Patriccollu, 2024-12-01:
@dennisguse Thanks a lot for your answer and precision about Weblate's comment.