OpenTracksApp/OpenTracks #2030: API CreateMarkerActivity ShowMarkerActivity App crashes when called

tch123, 2024-12-15:
API CreateMarkerActivity, ShowMarkerActivity App crashes when called
OpenTracks version 4.16
Android 14 LineageOS 21

dennisguse, 2024-12-16:
Do you have a stack trace?

tch123, 2024-12-17:
### App information
* ID: de.dennisguse.opentracks
* Version: 5941 v4.16.0

### Device information
* Brand: OnePlus
* Device: OP594DL1
* Model: CPH2447
* Id: AP2A.240905.003
* Product: CPH2449

### Firmware
* SDK: 34
* Release: 14
* Incremental: 7c78e8f134

### Cause of error
```java
Exception in thread "main": java.lang.RuntimeException: Unable to start activity ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.ui.markers.MarkerEditActivity}: java.lang.IllegalStateException: TrackId and Location must be provided or an existing markerId
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3993)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:4173)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:114)
	at android.app.servertransaction.TransactionExecutor.executeNonLifecycleItem(TransactionExecutor.java:231)
	at android.app.servertransaction.TransactionExecutor.executeTransactionItems(TransactionExecutor.java:152)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:93)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2595)
	at android.os.Handler.dispatchMessage(Handler.java:107)
	at android.os.Looper.loopOnce(Looper.java:232)
	at android.os.Looper.loop(Looper.java:317)
	at android.app.ActivityThread.main(ActivityThread.java:8592)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:580)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:878)
Caused by: java.lang.IllegalStateException: TrackId and Location must be provided or an existing markerId
	at de.dennisguse.opentracks.ui.markers.MarkerEditActivity.onCreate(MarkerEditActivity.java:101)
	at android.app.Activity.performCreate(Activity.java:8960)
	at android.app.Activity.performCreate(Activity.java:8938)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1526)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3975)
	... 13 more
```

dennisguse, 2024-12-17:
Thanks. 

Can you describe the steps to reproduce this?
And does this always happen?

tch123, 2024-12-18:
Steps to reproduce this situation:
1. Run OpenTracks and record the track
2. Use Tasker App to call OpenTracks API CreateMarkerActivity
> Tasker creates a new task>Start the application>Select the application OpenTracks first and long press to select CreateMarkerActivity from the pop-up menu
3. Crash occurs

This happens every time after following the above steps

dennisguse, 2024-12-18:
The crash is correct.
You need to include the proper parameters: location and track_id. 
And thelocation must even be an Android Location object. 
I believe you cannot provide this via Tasker. 

We can improve the API documentation, but the behavior will still be the same. Only the error message would be nicer. 

And this part of the API was meant for OSMDashboard. 
For other uses, we should likely rethink it. 

@pstorch FYI

dennisguse, 2024-12-19:
@tch123 by the way: what did you try to do?
What is your use case for using these (internal) APIs?

dopdo, 2024-12-19:
i try :
bug1:
### App information
* ID: de.dennisguse.opentracks
* Version: 5909 v4.14.1

### Device information
* Brand: Xiaomi
* Device: crux
* Model: Mi9 Pro 5G
* Id: RKQ1.200826.002
* Product: crux

### Firmware
* SDK: 30
* Release: 11
* Incremental: V13.0.1.0.RFXCNXM

### Cause of error
```java
Exception in thread "main": java.lang.RuntimeException: Unable to start activity ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.io.file.exporter.ExportActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.net.Uri.getPathSegments()' on a null object reference
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3540)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3700)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:85)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2136)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:236)
	at android.app.ActivityThread.main(ActivityThread.java:8060)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:656)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:967)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.net.Uri.getPathSegments()' on a null object reference
	at android.provider.DocumentsContract.getTreeDocumentId(DocumentsContract.java:1274)
	at androidx.documentfile.provider.DocumentFile.fromTreeUri(DocumentFile.java:136)
	at de.dennisguse.opentracks.io.file.exporter.ExportActivity.onCreate(ExportActivity.java:164)
	at android.app.Activity.performCreate(Activity.java:8143)
	at android.app.Activity.performCreate(Activity.java:8114)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1310)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3513)
	... 11 more
```
bug2:
### App information
* ID: de.dennisguse.opentracks
* Version: 5909 v4.14.1

### Device information
* Brand: Xiaomi
* Device: crux
* Model: Mi9 Pro 5G
* Id: RKQ1.200826.002
* Product: crux

### Firmware
* SDK: 30
* Release: 11
* Incremental: V13.0.1.0.RFXCNXM

### Cause of error
```java
Exception in thread "main": java.lang.RuntimeException: Unable to start activity ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.io.file.importer.ImportActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.net.Uri.getPathSegments()' on a null object reference
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3540)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3700)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:85)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2136)
	at android.os.Handler.dispatchMessage(Handler.java:106)
[... 195 lines cut ...]

### Cause of error
```java
Exception in thread "main": java.lang.RuntimeException: Unable to start activity ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.TrackStoppedActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'long de.dennisguse.opentracks.data.models.Track$Id.id()' on a null object reference
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3540)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3700)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:85)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2136)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:236)
	at android.app.ActivityThread.main(ActivityThread.java:8060)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:656)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:967)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'long de.dennisguse.opentracks.data.models.Track$Id.id()' on a null object reference
	at de.dennisguse.opentracks.data.ContentProviderUtils.getTrack(ContentProviderUtils.java:263)
	at de.dennisguse.opentracks.TrackStoppedActivity.onCreate(TrackStoppedActivity.java:48)
	at android.app.Activity.performCreate(Activity.java:8143)
	at android.app.Activity.performCreate(Activity.java:8114)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1310)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3513)
	... 11 more
```

tch123, 2024-12-19:
@dennisguse 
I used StartRecording and StopRecording together with Tasker. This can generate an icon on the desktop and start recording when clicked.

Use Tasker to get home (specify Wi-Fi and call StopRecording API to stop recording). This will prevent me from forgetting to stop recording.

If Tasker can directly call the mark location API, this operation can be bound to the volume keys, which is easy to operate while wearing gloves in the cold winter.

This is the usage scenario I have in mind.

dennisguse, 2024-12-19:
@tch123 that's a case the API was not meant to be used for :D 
What you need is `while recording, please create a marker here`.
Can you create a new issue for this?

@dopdo can you please create a new issue for this and also describe how to reproduce the issue.
i don't know what you did, besides you tried to import files.

tch123, 2024-12-19:
@dennisguse I've created a new issue for the feature request to add quick marking functionality via Tasker. Please take a look when you get a chance. Thank you!

dennisguse, 2024-12-19:
Follow-up is: #2031

closed 2024-12-19 by dennisguse