OpenTracksApp/OpenTracks #2031: Request to Add API for Quick Marking via Tasker

tch123, 2024-12-19:
# Description:

As a user, I would like to be able to quickly mark locations using Tasker, especially during recording. This would significantly enhance the convenience of operation during outdoor activities, particularly in the cold winter while wearing gloves.

# Feature Request:

## Functionality: 
Add an API or feature that allows Tasker to trigger a quick marking of the current location during recording.

## Use Case:
After starting a recording, users can use a Tasker shortcut (like volume keys) to quickly mark the current location.

## Expected Outcome:
Users should be able to create a marker at their current location during recording by calling the API through Tasker.
The marking operation should be simple and quick, capable of being triggered by physical buttons (like volume keys) to accommodate operations while wearing gloves.

# Additional Information:
The current version of the API does not seem to be designed for this use case, but this would be a very useful extension.
If possible, I hope this feature can be implemented soon to meet the needs of users during outdoor activities.
Thank you to the development team for considering this feature request!

dennisguse, 2024-12-19:
For the new marker:
1. Should there be additional data stored (available is already time and location as well as to which track it will belong)?
2. Should it be created (aka persisted to storage) or
3. Should the UI be opened, so it will be stored when the user presses (but he can add more data himself)?

dennisguse, 2024-12-19:
Notes to myself: 
* Marker creation is currently triggered by `TrackRecordingActivity` while persisting is handled by `MarkerEditActivity`
* `TrackRecordingService` is only used to fetch last known location
* PublicAPI (as of now) has only access to `TrackRecordingService`

tch123, 2024-12-19:
1. I believe the current data (time, location, and track affiliation) is sufficient to describe the marker in detail.

2. Given my use case involves quickly creating markers, I prefer the method of direct creation (the marker name could be 'Marker 1', 'Marker 2', etc., which can be edited later). However, if there's an option or a separate method to open the UI for editing, that would be even better.

dennisguse, 2024-12-19:
@tch123 Have fun playing around: https://github.com/OpenTracksApp/OpenTracks/pull/2032
Completely untested...

commit 4653bdfa (2024-12-19): Public API: allow marker creation.

Fixes #2031.
commit d531d5e9 (2024-12-21): Public API: allow marker creation.

Fixes #2031.
PR #2032 (2024-12-21): Public API: allow marker creation.
Fixes #2031.
commit 522d9ebe (2024-12-21): Public API: allow marker creation.

Fixes #2031.
closed 2024-12-21 by dennisguse