OpenTracksApp/OpenTracks #2035: No application supporting OpenTracks Dashboard API

Der-Flip, 2024-12-27:
I used to be able to share a track from OpenTracks directly to tracklia via the map icon at the bottom. It then gave me the option to share via gpx, kml etc and then I could chose my app (tracklia, and other apps). Since some time now it just shows the error there is no application supporting the OpenTracks API.

Is this something that changed in OpenTracks or is it something that may have changed in tracklia?

I wouldn't mind too much, unfortunately tracklia has never been shown in the share menu and still isn't.

dennisguse, 2024-12-27:
We removed this functionality has let to some user confusion (mostly related why OSMDashboard didn't open).
This functionality was removed in https://github.com/OpenTracksApp/OpenTracks/releases/tag/v4.15.0

OSMDashboard (as a real user of OpenTracks Dashboard API)  provides some advantages:
1. live updating
2. can open markers in OpenTracks
3. the KML/GPX integration is still available: you can use the share icon

Der-Flip, 2024-12-27:
Ah, shame. As mentioned unfortunately tracklia never appeared in the share-icon app selection (I know this is a tracklia issue).

I liked the comfort of stopping the recording and then seamlessly sending it to tracklia. I do have an auto export set up so I'll probably go via the file explorer, which has its own issues (can't multi select and send to tracklia, but that's another issue that's is fully unrelated with opentracks, so no worries about that :) )...

closed 2024-12-27 by dennisguse