OpenTracksApp/OpenTracks #2039: Possible improvement to the computation of *lap speed/pace*

NiaJ3oE2LM, 2024-12-29:
### Is your feature request related to a problem? Please describe.

I've tried to keep an eye on issues related to the computation of the lap statistics but I'm still not convinced by the way the app computes the *interval* (lap) speed/pace. Here is my feedback from field testing; assume you're moving at 5 min/km (12 km/h) and that GPS signal is sufficiently strong (for example, on an open field).

+ if you set `Time interval`= 1 minute and `Lap length`= 500m you will hear the *exact* same lap speed/pace twice since the app does not seem to recompute the statistics based on the *last* interval (500m) from the position when the announcement was triggered;
+ if you set `Time interval`= 1 minute and `Lap length`= 100m you will still hear the *exact* same statistics twice but the value will update more frequently.

### Describe the solution you'd like

Use the last `Lap length` meters from the position/time of the announcement trigger in order to compute a more meaningful lap speed/pace. The `Lap length` used for this computation should be independent from: 

+ the `length` that triggers a voice announcement
+ the `length` associated to GPS recording precision

### Additional context

May be related to:

+ [Interval Stats announced incorrectly #443 ](https://github.com/OpenTracksApp/OpenTracks/issues/443)
+ [Voice announcements: trigger on both time and distance frequency #921 ](https://github.com/OpenTracksApp/OpenTracks/issues/921)

App version `v4.16.1` installed from f-droid on android 10 and used for tracking **running** sessions.

dennisguse, 2024-12-29:
Any PR on this is welcome.

dennisguse, 2025-06-19:
@NiaJ3oE2LM would you mind to explain me the problem again?
I just was reading the code again and what happens is that the intervals are computed from the beginning of track.
That is fine for length based announcements.

What could be changed is for time-based announcements, compute the interval from "most recent position" aka from the end.

Does this make sense?

NiaJ3oE2LM, 2025-06-19:
Hello, yes certainly.

Since the Statistics announcements page only has the **average** (moving ?) and the **lap** speed/pace announcements (for speed/pace) I was suggesting that the **lap** speed/pace might be used as a more general *instantaneous*  speed/pace announcement triggered by a timer, and filtered with the `lap length` parameter.

Consider the case where: `lap length`= 1 km, (announcements) `time interval`= 1 minute, and assume you are moving at 5 min/km. 
In a 1 km run, what will be the 5 **lap** speed/pace announcements ?
I did not look into the code but I  guess the app will probably announce the same value 5 times (the value of the previous lap). 
The last announcement will probably be overlapped because the *lap* trigger coincides with the *time* interval trigger.

More generally, assuming that the **average** speed/pace announcement is the (total) run average speed/pace, it would be nice to have a third type of speed/pace called, for example, **interval** speed/pace. This new metric would:
+ compute the average speed on the last `interval length` meters (new parameter to be added in the settings page, shall be grater than the `recording distance interval`)
+ be triggered only by the `time interval` timer

In my opinion, the **interval** speed/pace would be helpful for *interval training* where you run at a certain pace for a certain time, then rest and repeat with possibly varying sprint lengths.

If this were to be implemented, three speed/pace announcements would become available:
1.  (existing) total run **average** speed/pace triggered by `time interval`;
2. (existing) **lap** speed/space computed on the `lap length` partition of the run - aka from the start- and triggered both by `time interval` and end of *lap*;
3. (new) **interval** speed/pace computed on the last `inerval length` - aka from the end - and triggered by `time interval`