OpenTracksApp/OpenTracks #2041: Running cadence computation from raw accelerometer recording

NiaJ3oE2LM, 2024-12-29:
### Is your feature request related to a problem? Please describe.
It would be super nice to be able to compute running cadence just by filtering raw accelerometers recorded data.

### Describe the solution you'd like

Assuming you wear an arm band or something similar, this should be possible by resorting to basic digital filtering techniques. 

Have you ever thought about it? How could I help implementing such a feature?

dennisguse, 2024-12-29:
You are looking for this: https://developer.android.com/health-and-fitness/guides/basic-fitness-app/read-step-count-data
And not really - it is likely too noisy.

dennisguse, 2024-12-29:
PS/ I use dedicated sensors for these cases...

closed 2025-02-16 by dennisguse