OpenTracksApp/OpenTracks #2043: Speed jumps to 0.0

Firesphere, 2025-01-01:
**Describe the bug**

As of recently, the speed shows on screen as 0.0 a lot. 
The speed jumps between the actual speed and 0.0, while the distance/altitude etc. measurements keep going as normal.

When importing in to FitTrackee, I don't see these drops in speed in the uncleaned charts generated, hence I'm _assuming_ it's a display issue and not an actual recording issue.

**To Reproduce**
1. Start a new track
2. Start moving
3. Keep an eye on the current speed
4. Notice the speed jumping to 0

If applicable:
* Can't really screenshot while cycling
* Would love to, if possible... How can I get a log output?

**Technical information**
 - Device: Oppo Find X2 Lite
 - OS: ColourOS V12.1
 - OpenTracks version: v4.17.0

dennisguse, 2025-01-02:
This should happen if you become idle. 
Can you check if the time to idle is too low in your case?

If OpenTracks becomes idle, there can be a voice announcement. 
This helps to identify it.

Firesphere, 2025-01-02:
It's definitively not while _me_ being idle that it jumps to 0, I was definitely multiple times heading downhill at a reasonable (30+km/h) speed!

Next time I go for a ride, I'll see if I can get some logging going, and/or video of it

Firesphere, 2025-01-03:
Here's a screen-capture of a few seconds of me riding.
https://static.firesphere.dev/Record_2025-01-03-14-22-38.mp4

I'm certain it's not bad GPS reception, as the tracks import fine in FitTrackee, with the correct locations, and speeds, hence my thought originally, that it's purely a visual/on-screen thing.

dennisguse, 2025-01-03:
Thanks
I have a gut feeling where I broke something...
Somewhere in the `reset()` part.

Can you upload the video directly to Github?
Then it is also available in the future.

dennisguse, 2025-01-03:
Likely are those two:
* 01f4359c
* 333bc6f3

Firesphere, 2025-01-03:
As per request :) 
https://github.com/user-attachments/assets/002bae37-bc7a-4918-98c8-237f1e4c71f3

Stephane-74, 2025-01-04:
Hello,
I have the same problem here. Sorry I couldn't make a screenshot, but I experienced it today. The speed stays at 0, but shows the value once n a while, for one or two seconds. Same thing with the altitude. Max speed, duration and time were OK.
Sorry I can't help more.

dennisguse, 2025-01-06:
Found the problem: it is a side effect of AggregatorGPS.reset() [trigger on new GPS data] and the TrackPointCreator.createCurrentTrackPoint()

No effect on recorded data; only UI.  
Just looks ugly. 

Will take some days to fix.

Stephane-74, 2025-01-06:
Thanks for the explanation and taking care of it.
OpenTracks is really a great ap, I'm happy to use it as soon as I'm on my bike.
Take care 💪

-------- Message d'origine --------

dennisguse, 2025-01-20:
Fixed by reverting: 7df625ff0c5504e37afd372bf9c1a458a96ee8b6

closed 2025-01-20 by dennisguse