OpenTracksApp/OpenTracks #2044: can't export map

mercier-beauregeard, 2025-01-02:
Since the new version it's impossible to share my track to a map ( like osmand+)
When I click on the button made for, I have an error message. "There no application supporting the API"

[image]**Technical information**
 - Device: [pixel 4a5g)
 - OS: [e.g. LineageOS 15.1]
 - OpenTracks version: 4.17.0

dennisguse, 2025-01-02:
This functionality got removed in v4.15.0
See for more details #2012

If this absolutely necessary it could be re-implemented using a separate button.

dennisguse, 2025-01-02:
PS/ I would recommend to use OSM Dashboard
https://f-droid.org/packages/de.storchp.opentracks.osmplugin/

closed 2025-01-02 by dennisguse

mercier-beauregeard, 2025-01-07:
Ok thank you, i'll try OSM Dashboard and it work perfectly, i just have to understand how to use diffrent map.

Thanks a lot