OpenTracksApp/OpenTracks #2045: Fastest pace calculation

linuzzz, 2025-01-02:
Hello,

just started using OpenTracks and in my first run I have a fastest pace of 4 min/km in the Stats tab.
Going into the intervals tabs I have a minimum of 5.44 min/km if I split by 1km or a minimum of 4.51 min/km if I split by 100m

So how could the fastest pace be 4 min/km? Is this some kind of instantaneous calculation?

Second question, if I may...is there a way to have the intervals tabs doing the pace calculation without the rest time?

I am using v4.16.1 on Android 14 on a Galaxy S22

Thanks

dennisguse, 2025-01-02:
Stats tab: the pace for the whole track (until now) [both moving and overall are available]
Intervals: depends on the interval distance (and it is the OVERALL pace; not the MOVING pace).

Anyhow, there will be some differences for different interval distances as new data is only stored every 10m (default).
So, the smaller the interval, the higher the noise aka variation.

Regarding your 2nd point: I don't know what you mean.

linuzzz, 2025-01-03:
thanks for your reply, so if I understood correctly it could be that I did 4min/km in a 10m distance but looking in intervals by 100m the fastest was anyway 4.51min/km?

about the second question it is about the option of having intervals taking into account only MOVING pace and not the OVERALL

thanks!

closed 2025-02-16 by dennisguse