OpenTracksApp/OpenTracks #2049: Visual bug: Bottom bar can overlap with UI elements

thomas-esau, 2025-01-11:
**Describe the bug**
The bottom bar (e.g. while in settings) that is used for some navigation or option buttons can burrow other UI elements. 

In this image (setting a marker), you can see, how this affects usability:
[image]

**To Reproduce - settings menu**
1. Go to: 'settings'
2. Click on 'Recording'
3. Scroll down to the bottom
4. See the option group 'Advanced' burrowed behind the bottom bar

**Technical information**
 - Device: Google Pixel 6a
 - OS: Android 15
 - OpenTracks version: v.4.17.5irreproducible/v.4.17.5 / Version code: 5989

As the visual behaviour is currently a little off due to Android 15 and targetSDK 35, it's not surprising this happens.

Related issues:

- https://github.com/OpenTracksApp/OpenTracks/issues/1955
- https://github.com/OpenTracksApp/OpenTracks/issues/2008

dennisguse, 2025-01-11:
Sadly, I cannot reproduce this: on my Android 13 it works and in a SDK35 emulator as well. 

Can you fix this yourself?

dennisguse, 2025-02-16:
@thomas-esau any more info?

dennisguse, 2025-02-22:
@DanyPM is this happening only in the settings or somewhere else as well?

bunkfunk44, 2025-02-25:
Same issue on Android 15 (Pixel 6). Just downloaded the app and noticed this in settings too - the very bottom of the pages are blocked by that "back" bar.

dennisguse, 2025-02-25:
I will check how it looks in the emulator for the Pixel 6. Thanks for the hint!

dennisguse, 2025-02-27:
In the settings, I can reproduce the problem, but I have no clue how to fix this :/

AndrejsK, 2025-03-04:
Poked around with it a bit, looks like the immediate issue is that the `LinearLayout` in `settings.xml` has `paddingBottom` set to `?attr/actionBarSize`, but the height of the `BottomAppBar` is `wrap_content`, which is sometimes actually larger than the padding. If you force `layout_height` of the `BottomAppBar` to `?attr/actionBarSize` it looks fine.

Just due to the size of the overlap I have a suspicion the core problem has to do with the bottom navigation bar - it looks like the height of the nav bar is added to the bottom app bar. As a disclaimer, this is the first time I've touched a native Android project, sorry if this is not helpful.

dennisguse, 2025-03-04:
Let's give this a try.
PS/ I was expecting that the UI components would take care about this.
For Jetpack compose, there are some things regarding padding, but I don't want to add Kotlin only for this stuff.

commit 5ae7ef35 (2025-03-06): Bugfix: bottombar overlapped content.
Also the FAB now looks Material3.

Fixes #2049.
commit 9f442c2e (2025-03-06): Bugfix: bottombar overlapped content.
Also the FAB now looks Material3.

Fixes #2049.
closed 2025-03-06 by dennisguse