OpenTracksApp/OpenTracks #2062: Request: Allow export of individual tracks or user-selected tracks

docgunthrop, 2025-02-21:
At the current moment, the only export option is all or nothing. I would like to suggest allowing the user the option to select individual tracks to export.

I had recently migrated from the older OpenTracks app to the newer reproducible build version 4.17.8. Once I successfully imported all my tracks to the new version, I deleted the older irreproducible build. Aside from a slight difference in aggregate stats (total miles, average speed, and maybe a couple other stats were slightly different from what showed in the other version, but by just a tiny percentage), everything else seems to work fine.

However, when I attempted to export my most recent track, this build began to export every single track again (several hundred in total). I aborted the process after it had already exported a few tracks, and then compared the existing GPX files (exported from the older version and imported into this new version) to those GPX files I had just exported from the new version and found that it was producing different file names, indicating that this new version is not recognizing these tracks as the same ones that were just imported.

On the other irreproducible version, when exporting, the app allowed me to select to skip any duplicates but that option did not show up this time. Rather than having to re-export every single track again, I'd like to suggest including the feature to manually select tracks to export.

dennisguse, 2025-02-21:
1. The share functionality may be helpful 
2. How are the filenames different ?
3. I would recommend exporting as KMZ as GPX has quite some limitations (if you are using additional sensors).

docgunthrop, 2025-02-23:
> The share functionality may be helpful

Unfortunately, the share function doesn't allow me to "send" the file to the file manager on my device.

> How are the filenames different ?

The ending characters are different for the same track. For example, a track I exported from the irreproducible build on 1/30 has the filename `2025-01-30_15_48_42.748.gpx`. I imported this track into the reproducible build, but when exporting from this build the filename produced is `2025-01-30_15_48_43.177.gpx`. This seems to be the reason why it's not detecting duplicates when running an export.

dennisguse, 2025-02-27:
TL;DR... that's a GPX situation - for export/import always go with KMZ.

What happens:
1. Filename contains the track start timestamp (as configured by default in settings)
2. On import OpenTracks, uses the timestamp of the first TrackPoint to determine the start timestamp
3. GPX only contains the TrackPoints with GPS coordinates (GPX does not support TrackPoints with GPS coordinates while KML does)
4. ... the first TrackPoint has NO GPS and is thus not exported.

In addition, OpenTracks generates TrackPoints without coordinates for
* idle
* non-GPS sensor data
* start, pause and stop events triggered by the user

So, all the statistics will be off when using GPX (at least compared to KML).

dennisguse, 2025-02-27:
Solution: delete all exports and re-create them as KML.
You will not be able to re-create the lost data, but it will be nicer next time.

closed 2025-02-27 by dennisguse