OpenTracksApp/OpenTracks #2071: App crashes when trying to delete a marker

rusty-snake, 2025-03-10:
**Describe the bug**
App crashes when trying to delete a marker.
Reproducibility: always

```
type: crash
osVersion: google/oriole/oriole:15/BP1A.250305.019/2025030800:user/release-keys
package: de.dennisguse.opentracks.playstore:6095, targetSdk 35
process: de.dennisguse.opentracks.playstore
processUptime: 7838 + 238 ms
installer: dev.imranr.obtainium

java.lang.NullPointerException: Attempt to invoke virtual method 'boolean java.io.File.exists()' on a null object reference
	at de.dennisguse.opentracks.data.ContentProviderUtils.deleteMarkerPhoto(ContentProviderUtils.java:519)
	at de.dennisguse.opentracks.data.ContentProviderUtils.deleteMarker(ContentProviderUtils.java:446)
	at de.dennisguse.opentracks.ui.markers.DeleteMarkerDialogFragment.lambda$onCreateDialog$0(DeleteMarkerDialogFragment.java:77)
	at de.dennisguse.opentracks.ui.markers.DeleteMarkerDialogFragment.$r8$lambda$eS5iEMpzIDwBNbBeXDEkifF19P4(Unknown Source:0)
	at de.dennisguse.opentracks.ui.markers.DeleteMarkerDialogFragment$$ExternalSyntheticLambda1.run(D8$$SyntheticClass:0)
	at java.lang.Thread.run(Thread.java:1119)
```

dennisguse, 2025-03-10:
Does this happen for an existing marker (before te update) with an attached picture?

Or for a new markers/o picture?

rusty-snake, 2025-03-11:
Existing marker without picture (I can not test with picture).

Not for new markers.

dennisguse, 2025-03-14:
Are you able to get a logcat?
I have a feeling why this might happen, but it's just a feeling.

rusty-snake, 2025-03-14:
I want to review the full logcat before posting. However this part looks interesting:

```
W BitmapDrawable: BitmapDrawable created with null Bitmap
E MarkerListAdapter: Failed to image java.io.FileNotFoundException: No content provider: 
W BitmapDrawable: BitmapDrawable created with null Bitmap
W BitmapDrawable: BitmapDrawable created with null Bitmap
E MarkerListAdapter: Failed to image java.io.FileNotFoundException: No content provider: 
W BitmapDrawable: BitmapDrawable created with null Bitmap
E MarkerListAdapter: Failed to image java.io.FileNotFoundException: No content provider: 
```

because it reflects how the UI looks like for old markers. They have the same gradient for as markers with pictures in the list and no orange pin when opened. And indeed, if I edit them and add a picture, I can delete them.

dennisguse, 2025-03-14:
Let me be specific: `Log.w(TAG, "External photo contains no filename.");`.
Is this message happening right before?

I probably should refactor this to just throw an exception and crash...

rusty-snake, 2025-03-14:
> Let me be specific: Log.w(TAG, "External photo contains no filename.");.
Is this message happening right before?

Yes. Last message before FATAL EXCEPTION.

commit 07448439 (2025-03-15): Bugfix: Markers without photo could not be deleted (created pre v4.18.0).

Introduced in 1b33184835cb15ba2b598116ce8589ed320dd42b

Fixes #2071.
closed 2025-03-15 by dennisguse