OpenTracksApp/OpenTracks #2073: BottomAppBar partly hidden: Pixel 8 Pro / Android 15 [v4.18.0]

dennisguse, 2025-03-13:
With the release of v4.18.0, the BottomAppBar was manually set in height.
However, on a Pixel 8 Pro in vertical mode (with high resolution setting), the BottomAppBar partly hidden.
Landscape works.
This effect only occurs if a FAB Button (start/ stop recording) is shown.

The problem cannot be reproduced on:
* Pixel 8 with Android 15
* Pixel 8 Pro Emulator

[image]
[image]
[image]
[image]
[image]

https://github.com/OpenTracksApp/OpenTracks/commit/5ae7ef350c6693080865fcb64bd4eb202701f5fe

dennisguse, 2025-03-13:
The problem seems to be unrelated to the full resolution mode of the Pixel 8 Pro.

Buntbart, 2025-04-18:
same Problem here with v.4.19.0 from F-Droid on a Oneplus 6t under LineageOS 22.1 (Android 15).

[image]

[image]

Megadash452, 2025-04-18:
Same issue on a galaxy S24 with oneui7/android 15. It also happens on screens without the FAB, as shown in the picture.

[image]

Megadash452, 2025-06-04:
@dennisguse According to [this article](https://developer.android.com/develop/ui/views/layout/edge-to-edge), Android forces apps targeting SDK >=35 (aka Android 15 or greater) to render with the full screen area, causing them to render behind the status and nav bars, even if the app did not call [androidx.activity.ComponentActivity.enableEdgeToEdge](https://developer.android.com/reference/androidx/activity/ComponentActivity#(androidx.activity.ComponentActivity).enableEdgeToEdge(androidx.activity.SystemBarStyle,androidx.activity.SystemBarStyle)).

I tried changing the app to target SDK 34 instead, and it seems to run fine (though I haven't tried it on a real phone with gps and sensors). Would reverting to SDK 34 be an acceptable solution?

dennisguse, 2025-06-04:
@Megadash452 yes, the problem is the full screen approach.
Downgrading would need to get down to 33 IIRC.
However, this is not a solution - just a very hot fix.

In any case, the bottom navbar (as it is a default material component) should just behave accordingly, but seemingly it doesn't.
Fun aspect: I cannot reproduce this on Android 13 and "luckily" I didn't get an upgrade to Android 14 so far.

If you want to take a look and play around: it is the `BottomAppBar` in track_list.xml`.
Other activities will be affected as well, but that is the most prominent one.

Buntbart, 2025-06-04:
There was a similiar (if not the same) problem in https://github.com/mueller-ma/PrepaidBalance/issues/309
Maybe you would like to ask the developer about his solution.

Megadash452, 2025-06-04:
> I cannot reproduce this on Android 13 and "luckily" I didn't get an upgrade to Android 14 so far.

This problem was not occurring in Android 14, only Android 15.

> If you want to take a look and play around: it is the `BottomAppBar` in track_list.xml`. Other activities will be affected as well, but that is the most prominent one.

I will later today, though I have no experience with Views so it might take me a while. Thanks 👍

dathoo, 2025-06-19:
Same issue on Oppo Find X3 Neo 5G - Android 13.

In the "non-reproducible" version of the app, this display bug doesn’t occur.

commit 5bbb5f78 (2025-06-20): Bugfix: apply edge-2-edge insets to everything always.

Fixes #2073.
closed 2025-06-20 by dennisguse