OpenTracksApp/OpenTracks #2074: GPX format when imported into Garmin Connect doesn't show cadence or speed from sensor

TheSkorm, 2025-03-14:
**Describe the bug**
GPX format when imported into Garmin Connect directly is missing sensor data. Additionally the speed shown by Garmin is very noisy, and the activity type is set to "Other"

[image]

**To Reproduce**
1. Record a bike ride with speed and cadence
2. Export activity as GPX
3. Import into Garmin Connect. Sensor data will be missing.

[image]

Additionally using http://gotoes.org/strava/Combine_GPX_TCX_FIT_Files.php to convert GPX to FIT the speed and cadence data does show up, and the activity type is set.

[image]

**Technical information**
 - Device: Pixel 8A
 - OS: Lineage
 - OpenTracks version: v4.18.0

dennisguse, 2025-03-14:
OpenTracks uses GPX TrackPointExtensionv2 to export this kind of data.
See: https://www8.garmin.com/xmlschemas/TrackPointExtensionv2.xsd

I suggests to contact Garmin Connect people about this.

TheSkorm, 2025-03-15:
Hi @dennisguse,

I think you could consider looking into this issue again. The output of the GPX file from opentracks appears very badly formatted - probably caused generating XML using string manipulation. 

Some notes:
- Heart rate shouldn't be included if its less than 1 `<xsd:minInclusive value="1"/>`
- `TrackPointExtension` are a sequence which means they need to be in the correct order (weird decision, but that's what it is). So `gpxtpx:speed` needs to be last
- `opentracks` and `cluetrust` aren't part of `TrackPointExtension` so should either be under `trkpt->extensions` or `trkpt->extensions->TrackPointExtension->Extensions`
- xsi:schemaLocation is invalid - `TrackStatsExtension` is missing the xsd file, opentracks schema 404s, http://www.cluetrust.com/Schemas/gpxdata10.xsd is now https://www.cluetrust.com/Schemas/gpxdata10.xsd
- TrackPointExtension v2 doesn't seem well supported, however TrackPointExtension v1 is. It does however drop the speed field. I'd suggest providing both TrackPointExtension v1 and v2.

The zip file contains a hand modified gpx file that implements some of these changes.

Below screenshots demonstrate that both in Strava and Garmin Connect see improvements with these changes. In Garmin Connect we get the heart rate data, and per lap bike and overall cadence stats. In Strava we see that the speed sensor is now used.

Unfortunately Garmin Connect doesn't seem to import all the cadence data still, and this appears to be an open [Garmin Connect issue](https://forums.garmin.com/apps-software/mobile-apps-web/f/garmin-connect-web/327244/garmin-connect-didn-t-import-the-cadance-from-my-gpsmap64s-activity). It also appears that Garmin Connect doesn't have a way of using the type field in the GPX to detect the activity type.

Ultimately it would be better if TCX support was added (https://github.com/OpenTracksApp/OpenTracks/issues/637) as it has much better support for health/activity data. GPX isn't really fit for purpose and especially falls apart during indoor activities.

Original file:

[image]
[image]

Modified:

[image]
[image]

GPX files:
[gpx.zip](https://github.com/user-attachments/files/19257595/Archive.zip)

dennisguse, 2025-03-15:
@TheSkorm thanks for looking into this.
And yes, we can make those changes.

dennisguse, 2025-03-16:
Fixed. Will be released as v4.19.0 today.

closed 2025-03-18 by dennisguse