OpenTracksApp/OpenTracks #2077: Integration of weather data

iointerrupt, 2025-03-15:
**Is your feature request related to a problem? Please describe.**
Add weather related data per track (start and end) - Temperature, Humidity, Condition (cloudy, sunny, ...) for each track if its outdoor related.

**Describe the solution you'd like**
Integrate with other open source weather related apps: Currently apps like  "Breezy Weather" or "QuickWeather" are FOSS projects on f-droid that integrate with GadgetBridge to allow it to send weather related data to wearables. Perhaps a collaboration to allow OpenTracks to pull that data from those apps. 

**Describe alternatives you've considered**
Determine relevant weather data points and extend the tracks table with those columns.  This way if we can find a way to pull historical weather related data for the start and end points at that specific time and place, that could be filled in for past runs.

dennisguse, 2025-03-22:
What it is the problem you are trying to solve?
Do you need the weather info during your outdoor activity or afterwards?

iointerrupt, 2025-03-25:
As an avid runner, the temperature can affect your pace.  Would be nice to be able to compare performance based on temperature, humidity, condition.  You do not need to monitor weather condition during runs, just the general consider at beginning and/or end of run.

closed 2025-06-15 by dennisguse