OpenTracksApp/OpenTracks #2081: Integration with Organic Maps (as an alternative to OSMdashboard)

uselessgames, 2025-03-23:
**Is your feature request related to a problem? Please describe.**
I already use Organic Maps, which has the same principles as Open tracks, and would be nice to use it, Vs having to install another mapping app. 

**Describe the solution you'd like**
To my knowledge; Current process opentracks only supports osmdashboard for viewing on a map. Would be nice if it also integrated with organic maps https://github.com/organicmaps/organicmaps

**Describe alternatives you've considered**
N/A

**Additional context**
Thank you for proving a great app that is stable and puts the user in charge of their data.

dennisguse, 2025-03-23:
@uselessgames have you checked with the people behind OrganicMaps?

uselessgames, 2025-03-23:
> have you checked with the people behind OrganicMaps?

No. I thought the flow would begin here. 

I can though;

Are there any requirements from this side for it to make it work?

uselessgames, 2025-03-23:
Oke. Thanks. 
I'm not developer so not fully understand. I'll try to make a suggestion on that side with the given example and docs. 
Cheers

closed 2025-03-23 by uselessgames

dennisguse, 2025-03-25:
@uselessgames you know, this is not a technical problem.
It can just be done and the integration shouldn't be too complicated.
The question is if the "user value" is sufficient enough to build this feature and keep it alive.
So, if enough people ask for it, the chance that it will be build increases :)