OpenTracksApp/OpenTracks #2085: Elevation recorded wrong

iameru, 2025-03-31:
I found out that `OpenTracks` seems to show/calculate the elevation wrongly.
I regularily find recorded biking sessions with elevations of 800 to 1100 meters. This cannot be true.
I confirmed this by recording a Track today with OpenTracks and with a Sportswatch and comparing with maps.openrouteservice.org and common sense.

I'll attach graphs made of the exported `.gpx` tracks, displayed by `viking` and by `fittrackee`. In the `OpenTracks` app itself the data looks like this aswell so it is not an export bug.

**Technical information**
 - Device: Google Pixel 4a 5G
 - OS: GrapheneOS 
 - OpenTracks version: v4.19.0 Irreproducable

Whats the elevation datasource of OpenTracks?

On a side note: Is there any way to fix my `.gpx` exports from the last years?

Here are Screenshots showing the huge difference in the aggregated elevation and in the elevation profile

[image]
[image]
[image]

dennisguse, 2025-04-01:
Altitude (as WGS84) is taking directly from the GPS of your device.
This is also how the data is stored internally as well as exported.
For the UI only, the altitude is converted to EGM2008 - so it represents "above sea level".

The only source of this inaccuracy may be the reception or the GPS receiver.
You know: trees, mountains, bad weather or keeping the phone in a backpack.

What you can do is to get altitude from DEM data (using lat/lng): https://www.gpsvisualizer.com/tutorials/elevation_gain.php

iameru, 2025-04-08:
Thanks for explaining 👍 
My data is from 2 devices over the last 3 Years with 3 iterations of different OS and all data is equally wrong. So please excuse me being still doubtful about the device GPS being the source of the huge error margin here.

I will record some tracks with osmand and other apps to be sure that this is the Phones GPS.

Thanks for the hint about how to correct the data!

dennisguse, 2025-04-08:
@iameru let me know what you find...

iameru, 2025-04-14:
I found confusion :)
You are, of course, right. While Osmand seems to have the correct elevation profile while routing for a navigation, its elevation explodes just like the one from OpenTrack when recording a track!

As you pointed out, this seems to be an issue of the GPS in the device. OpenTracks does not correct the GPS data elevation and it is probably out of scope.

I wish I knew earlier though. I am not sure how OpenTrack can care or educate us users here though.
thanks again

closed 2025-04-14 by iameru