OpenTracksApp/OpenTracks #2089: No cadence from Wahoo Blue SC

ImaginaryResources, 2025-04-10:
**Please describe the problem and how to reproduce it:**

Recently bought a Wahoo Blue SC because docs said it worked with both cadence and speed. I've confirmed it works with the Wahoo app.

First issue is that I'm unable to connect both cadence and speed to the app. Unless, I somehow, connect both quickly at the same time. Even in this scenario cadence value is set to 0.

I looked into https://github.com/OpenTracksApp/OpenTracks/issues/1050 and it seems like data is the same?

```
CSC Measurement
0x2A5B
Properties: NOTIFY
Parsed value: Wheel rev: 7197,
Last wheel event time: 60377 ms,
Crank rev: 2764,
Last crank event time: 59577 ms
Raw value: (0x) 03-1D-1C-00-00-D9-EB-CC-0A-B9-E8
```

Unsure if there's a need for an additional option "Distance, Speed, and Cadence" for cycling, and this unique sensor.

**Checklist**

- [ ] Consulted [https://dontkillmyapp.com/]
- [ x] Power management: battery optimization is disabled for OpenTracks

**Technical information**

- Device: Google Pixel 7
- OS: Grapheme OS
- OpenTracks version: 4.19.0

dennisguse, 2025-04-10:
This sensor is somehow special :)
It sends the data in one data packet (as this is also specified in the Bluetooth specification).
In OpenTracks, you have to select it as Distance+Speed as well as Cadence sensor in the settings.
Internally, OpenTracks pretends that it connects to two different sensors - luckily Android's Bluetooth stack takes care of this.

I have the same sensor at my bike and it works: Disance+Speed is fine while cadence is kind of shaky.
For cadence, I see that it misses quite some rotations.

So, technically it is a "works for me" situation.

ImaginaryResources, 2025-04-10:
Oh, that's good to know that you have the same sensor!

So, because I was testing. I removed the sensor from cadence, after I was able to add it with the "glitch" of adding both the cadence and distance/speed sensor. 

Perhaps my issue is not with the data, but connectivity? How do I properly add my sensor to both cadence and distance/speed? I can probably grab a screen recording if that helps.

ImaginaryResources, 2025-04-10:
Well of course, now that I try to capture its odd behavior it works flawlessly. 😂
I really tried getting this to work for an hour or two throughout the week.

If by chance I see this behavior occur again I'll reopen this bug.